openwrt/luci #2868: luci-theme-rosy: MAC-Address & IPv4-Address drop down boxes are not visible for the last entries on Network / DHCP and DNS

fantom-x, 2019-07-14:
OpenWRT: 19.07-SNAPSHOT

Latest Firefox on MacOS: when adding a new entry to the Static Leases list, the drop down boxes are hidden for MAC & IPv4 are hidden and the only way to use them is to switch the Bootstrap them. My Static leases list is ~50 entries long.
Those drop downs are only fully visible starting for the five row from the bottom.

@rosysong

cc5771027, 2019-07-19:
Can you provide a screenshot?

fantom-x, 2019-07-19:
The drop down boxes are covered by the white border.

[image]
[image]

rosysong, 2019-07-23:
@fantom-x  And please tell me your firefox version ? Or update your firefox browser to latest and have a try again.

fantom-x, 2019-07-23:
68.0.1; there is no later version than this.

CCnut, 2019-07-24:
what's your luci-theme-rosy version?

rosysong, 2019-07-24:
@fantom-x  It seems you should bump your source code to latest 19.07 commit.

fantom-x, 2019-07-24:
luci-theme-rosy - git-19.190.55584-ab90938-1

OpenWrt 19.07-SNAPSHOT r10269-5100629e32 / LuCI openwrt-19.07 branch (git-19.190.55584-ab90938) 

This is from the build from scratch a few hours ago. The issue is still there. Are you saying that it is not present for you?

fantom-x, 2019-07-24:
It is not happening in Chrome: the bottom whit band drops lower and I can see a few lines of the list and can scroll it.

rosysong, 2019-07-29:
@fantom-x  @CCnut Strange, below is the screenshot from our firefox browser(68.0.1) on 19.07 branch.

[image]
[image]

fantom-x, 2019-07-29:
Try editing in-place. I think you pressed the **Edit** button here. Instead, go back to the list of static IPs and click on the drop down box on the last row.

fantom-x, 2019-07-29:
Interesting, I do not see the **Edit** button in my 19.07 firmware; only **Delete** is visible on the right side. Either way, click on the arrow to the right of each drop down box on the last row.

It is hard to tell, it looks like the drop down boxes are not present on the screenshots you provided. Is 19.07 build pulling the right source?

[image]

fantom-x, 2019-07-29:
My screenshot above is showing the arrows on the right side of the IP drop down box. It is the one that needs to be clicked on on the last row and the dropdown box is hidden under the white ribbon.

rosysong, 2019-07-30:
@fantom-x  Sorry, there are no drop down box now, only popup window.

rosysong, 2019-07-30:
> luci-theme-rosy - git-19.190.55584-ab90938-1
> 
> OpenWrt 19.07-SNAPSHOT r10269-5100629e32 / LuCI openwrt-19.07 branch (git-19.190.55584-ab90938)
> 
> This is from the build from scratch a few hours ago. The issue is still there. Are you saying that it is not present for you?

```
git pull --rebase
 ./scripts/feeds update -a
 ./scripts/feeds install -a
rm -rf bin
make clean
make
```

jow-, 2019-07-30:
The LuCI openwrt-19.07 branch differs from master.

fantom-x, 2019-07-30:
If LuCI is different in 19.07 and master, then that explains the different behaviour. This is really a cosmetic issue and an annoyance. I can always switch back to Bootstrap or edit the file directly.

hnyman, 2020-12-29:
The broken theme has been removed.

closed 2020-12-29 by hnyman