openwrt/luci #2870: interfaces: no default route option for static protocol

ahmar20, 2019-07-15:
I would like to point attention to a simple thing that there is no `Use Default Gateway` option available in static protocol interfaces. The option is available in DHCP enabled interfaces but not in static ones. Of course one can manually edit the config files and set `option defaultroute '0'` but LuCI should provide this option there becasue not all the static interfaces are supposed to use the default routing table. Thank you.

jow-, 2019-07-15:
This option makes little sense for static protocols. If you do not want to have default route on static interfaces, then don't set a gateway.

closed 2019-07-15 by jow-

Brain2000, 2023-01-04:
At the very least this option should be disabled by default for any interface protocol except ones that are autoconfigured (i.e. DHCP, DHCPv6).
The wording should also be "Create Default Gateway" as opposed to "Use Default Gateway". Or perhaps "Use As Default Gateway".

The words "Use Default Gateway" is misleading.

[You may have noticed I'm rather angry about this because it caused me hours of troubleshooting]