openwrt/luci #2871: luci-app-firewall Port forward rules are using list instead of option

luigi311, 2019-07-15:
When creating a port forwarding rule through luci it creates the configuration in  /etc/config/firewall using list instead of option such as for dest and src. Due to this the firewall is throwing out warnings and not accepting those rules so it is silently failing and the port is not being opened to the outside world. 

> config redirect
        option dest_port '32412'
        option proto 'tcp udp'
        option name 'test'
        option src_dport '32412'
        list dest 'lan'
        option dest_ip '<IP-ADDRESS>'
        list src 'wan'

> root@OpenWrt:~# /etc/init.d/firewall restart
Warning: Option @redirect[3].dest must not be a list
Warning: Option @redirect[3].src must not be a list
Warning: Section @redirect[3] (test) skipped due to invalid options

When i open up the config and change it manually to options instead of list doing the restart works and the warnings are gone. I am able to access the ports outside my network and everything is working perfectly.

teqlee, 2019-07-18:
Same problem with 19.07-snapshot.

AndreaBorgia-Abo, 2019-07-18:
@luigi311 this doesn't happen in 18.06.4 , right?
(luci-app-firewall version is git-19.170.32094-4d6d8bc-1)

closed 2019-07-19 by jow-

luigi311, 2019-07-25:
@AndreaBorgia-Abo This is occuring on OpenWrt SNAPSHOT r10504-ea44ed5f38 / LuCI Master (git-19.194.40998-7b1aabc)

commit dc021180 (2019-07-19): luci-base: ui.js: fix multi vs. multiple keyword confusion

The "luci.form" and "luci.tools.widgets" classes use the "multiple"
keyword while ui.js uses "multi" internally, leading to single-value
dropdowns widget values getting stored as uci lists.

Fix the issue by using "multiple" everywhere.

Fixes: #2871