openwrt/luci #2873: Network -> Interfaces 500 Internal Server Error

kfur, 2019-07-16:
OpenWrt 18.06.4 r7808-ef686b7292 / LuCI openwrt-18.06 branch (git-19.190.55614-35357e4)

I got a 500 Internal Server Error on the Network -> Interfaces with the error Message:

```Failed to execute arcombine dispatcher target for entry '/admin/network/network'.
The called action terminated with an exception:
/usr/lib/lua/luci/template.lua:97: Failed to execute template 'cbi/map'.
A runtime error occured: /usr/lib/lua/luci/template.lua:97: Failed to execute template '[string]'.
A runtime error occured: /usr/lib/lua/luci/model/firewall.lua:480: bad argument #4 to '?' (integer expected, got number)
stack traceback:
	[C]: in function 'error'
	/usr/lib/lua/luci/util.lua:41: in function 
	/usr/lib/lua/luci/model/firewall.lua:480: in function 'get_color'
	[string "[string]"]:4: in main chunk
stack traceback:
	[C]: in function 'error'
	/usr/lib/lua/luci/template.lua:97: in function 'render'
	...usr/lib/lua/luci/model/cbi/admin_network/network.lua:117: in function 'render'
	/usr/lib/lua/luci/cbi.lua:266: in function 'render_children'
	[string "/usr/lib/lua/luci/view/cbi/map.htm"]:1: in main chunk
stack traceback:
	[C]: in function 'error'
	/usr/lib/lua/luci/template.lua:97: in function 
	(tail call): ?
	/usr/lib/lua/luci/cbi.lua:257: in function 'render'
	/usr/lib/lua/luci/cbi.lua:440: in function 'render'
	/usr/lib/lua/luci/dispatcher.lua:923: in function 'target'
	/usr/lib/lua/luci/dispatcher.lua:959: in function```

jow-, 2019-07-19:
Can you please provide the output of `uci show firewall | grep name`?

Also what router model (architecture) is it? Do you run a self built firmware or an official image?

kfur, 2019-07-19:
```firewall.@zone[0].name='lan'
firewall.@zone[1].name='wan'
firewall.@rule[0].name='Allow-DHCP-Renew'
firewall.@rule[1].name='Allow-Ping'
firewall.@rule[2].name='Allow-IGMP'
firewall.@rule[3].name='Allow-DHCPv6'
firewall.@rule[4].name='Allow-MLD'
firewall.@rule[5].name='Allow-ICMPv6-Input'
firewall.@rule[6].name='Allow-ICMPv6-Forward'
firewall.@rule[7].name='Allow-IPSec-ESP'
firewall.@rule[8].name='Allow-ISAKMP'
firewall.guest_zone.name='guest'
firewall.guest_rule_dns.name='Allow DNS Queries'
firewall.guest_rule_dhcp.name='Allow DHCP request'
firewall.@zone[3].name='nanostation'
firewall.@rule[11].name='VLAN_ETH4 DHCP'
firewall.@rule[12].name='VLAN_ETH4 DNS'
```

ok maybe i shouldn't create issue because i'm using custom build for xiaomi mi router 3.

already fixed it by replacing `return "#%02x%02x%02x" % { 0xFF - r, 0xFF - g, 0xFF - b }` to `return "#eeeeee"`

closed 2019-07-19 by kfur