openwrt/luci #2881: Luci graphical issues with chrome

MSC2016, 2019-07-18:
The best way is to show what i am talking about:
[image]

Is anybody else having problems lik this?

lynksys wrt32x
Powered by LuCI openwrt-18.06 branch (git-19.170.32094-4d6d8bc) / OpenWrt 18.06.4 r7808-ef686b7292

jow-, 2019-07-18:
Already fixed with https://github.com/openwrt/luci/commit/9c5162d5967a56f2ca1ed6e84169c1b96b35d3ae

You cna run `opkg update; opkg upgrade luci-theme-bootstrap` to resolve this glitch.

closed 2019-07-18 by jow-

MSC2016, 2019-07-18:
LOL, thank you for your very fast response, you litteraly took less than 20 seconds to answer. SOLVED...