openwrt/luci #2885: error : /usr/lib/lua/luci/dispatcher.lua:247: /etc/config/luci seems to be corrupt, unable to find section 'main'

a7mag3ddon, 2019-07-19:
I have a Netgear R8000 I have installed  openwrt-bcm53xx-netgear-r8000-squashfs.chk 

I have then done the commands on ssh

opkg update
opkg install luci
/etc/init.d/uhttpd start
/etc/init.d/uhttpd enable 

To install luci however when i connect to the IP i get

/usr/lib/lua/luci/dispatcher.lua:247: /etc/config/luci seems to be corrupt, unable to find section 'main'
stack traceback:
	[C]: in function 'assert'
	/usr/lib/lua/luci/dispatcher.lua:247: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:144: in function </usr/lib/lua/luci/dispatcher.lua:143>

I dont seem be able to the next step which is to install the full WRT file.

From a bit of searching this has been a problem on and off but all the fixes i have tried either dont work or the page with the config i need to wget is no longer there.

Is this a bug? is there something i can do?

Many Thanks.

jow-, 2019-07-19:
Make sure that `rpcd` is running and enabled.

a7mag3ddon, 2019-07-19:
As far as i know it is, i did an  /etc/init.d/rpcd restart and got no errors

On thing i did notice was there was no luci in /etc/config so i after some reason i copied a backup and renamed it to luci, set chmod on the file to 777 and then use vi to check the contects which are as follows.

config core 'main'
        option lang 'auto'
        option mediaurlbase '/luci-static/bootstrap'
        option resourcebase '/luci-static/resources'

config extern 'flash_keep'
        option uci '/etc/config/'
        option dropbear '/etc/dropbear/'
        option openvpn '/etc/openvpn/'
        option passwd '/etc/passwd'
        option opkg '/etc/opkg.conf'
        option firewall '/etc/firewall.user'
        option uploads '/lib/uci/upload/'

config internal 'languages'

config internal 'sauth'
        option sessionpath '/tmp/luci-sessions'
        option sessiontime '3600'

config internal 'ccache'
        option enable '1'

config internal 'themes'
        option Bootstrap '/luci-static/bootstrap'

config internal 'apply'
        option rollback '30'
        option holdoff '4'
        option timeout '5'
        option display '1.5'

config internal 'diag'
        option dns 'openwrt.org'
        option ping 'openwrt.org'
        option route 'openwrt.org'

a7mag3ddon, 2019-07-19:
Is it possible to flash a different firmware directly from within the device?

Im still struggling tonget web gui to load up.

dartraiden, 2019-07-25:
> Is it possible to flash a different firmware directly from within the device?

[sysupgrade](https://openwrt.org/docs/guide-user/installation/sysupgrade.cli)

Ceremony64, 2019-07-28:
I am running into the same issue with snapshot r10250-016d1eb for the Archer C7 v2.
The current release 18.06.4 works, which I reverted back to.

Ceremony64, 2019-08-04:
Looking for a solution, rpcd isn't running on my router, as `pidof rpcd` returns nothing.

(Re)starting rpcd via `service rpcd` or `/etc/init.d/rpcd restart` doesn't seem to do anything (no error, but not running service either) and just trying to start rpcd directly gets me "segmentation fault"... I am not sure how to fix the underlying rpcd issue

a7mag3ddon, 2019-08-05:
Thanks for the update,  i gave up and serial flashed mine so all good here.

I wasnt skilled enough to work out what was wrong.

closed 2019-08-11 by jow-

mh4nx7net, 2020-06-17:
### **caution!! : upgrading with keep previous configuration cause some missing/system failure**
**so,**
**just upgrade with remove previous/ use default configuration**
by `sysupgrade` just add `-n` as your argument

**GET file.bin**
> if you still have remote access [ssh/telnet]
save file `*.bin` on temporary `/tmp` folder
**by netcat :**
ro: `opkg update && opkg install netcat`
ro: `netcat -lp 4444 > /tmp/file1.bin`
cl: `cat sys.bin | pv -b | nc [ro_addr] 4444 -w 3`
**by scp/sftp :**
cl: `scp sys.bin root@[ro_addr]:/tmp/.`

**DO sysupgrade -n -v**
> `ll /tmp/*.bin` if already exist
ro: `sysupgrade -v -n /tmp/*.bin`

hope help,
 [furthermore-keep read documentation carefully](https://openwrt.org/docs/guide-user/installation/sysupgrade.cli)