openwrt/luci #2887: luci-theme-rosy: Network / Firewall / Zones is cut off and can not be scrolled horizontally

fantom-x, 2019-07-19:
OpenWRT: 19.07-SNAPSHOT

Latest Firefox on MacOS: there is more information to the right of the last column, but it is hidden.

@rosysong

[image]

rosysong, 2019-07-24:
@fantom-x  please try this patch https://github.com/openwrt/luci/pull/2913

fantom-x, 2019-07-24:
It is now looking like this. A little messier, but all information is visible.

[image]

rosysong, 2019-07-29:
@fantom-x Would you please close this issue if it was fixed, thanks!

fantom-x, 2019-07-29:
I have just re-spun a new firmware image (19.07) by using the image builder and the issue is still present. It is either the package is not rebuilt yet, or 19.07 is not pulling the right sources code.

Yeah, looks like it is five day old at this time.

[image]

hnyman, 2020-12-29:
The broken theme has been removed.

commit 1b569350 (2019-07-26): luci-theme-rosy: fix #2887
commit 94f42488 (2019-07-26): Merge pull request #2913 from rosysong/openwrt-19.07

luci-theme-rosy: fix #2887
closed 2020-12-29 by hnyman