openwrt/luci #2888: luci-ssl-openssl/luci-app-statistics: Missing dependencies

fantom-x, 2019-07-20:
19.07-SNAPSHOT

@hnyman 

Using the image builder to build an image for r7800 and getting the following errors: are these dependencies being built? All the other packages I use are available, but these two are having issues.

```
Collected errors:
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for luci-ssl-openssl:
 * 	libip4tc2
 * 	libip6tc2
 * opkg_install_cmd: Cannot install package luci-ssl-openssl.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for luci-app-statistics:
 * 	libip4tc2
 * opkg_install_cmd: Cannot install package luci-app-statistics.
 * opkg_install_cmd: Cannot install package libip4tc2.
 * opkg_install_cmd: Cannot install package libip6tc2.
Makefile:152: recipe for target 'package_install' failed
make[2]: *** [package_install] Error 255
Makefile:111: recipe for target '_call_image' failed
make[1]: *** [_call_image] Error 2
Makefile:195: recipe for target 'image' failed
make: *** [image] Error 2
```

fantom-x, 2019-07-20:
It all works fine if I build the firmware from sources and those two packages are installed:

```
opkg list-installed | egrep "libip4tc2|libip6tc2"
libip4tc2 - 1.8.3-1
libip6tc2 - 1.8.3-1

```

hnyman, 2019-07-20:
Unrelated to LuCI.
There is no proper 19.07 buildbot yet, so I am surprised that you somehow found the image builder, as there are no downloadable packages for it.

The error itself is about iptables version bump from 1.8.2 to 1.8.3. part of packages are built along the kernel in phase1  buildbot and most in phase2. 

I wrote to you about this 17 days ago in https://forum.openwrt.org/t/errors-when-using-image-builder-for-ipq806x/39992/9?u=hnyman

The 19.07 branch is not yet fully usable, except with full source compilation.

closed 2019-07-20 by hnyman

hnyman, 2019-07-20:
Actually, looks like now is now finalising the 19.07 build infra...
https://buildbot.openwrt.org/openwrt-19.07/images/one_line_per_build
https://buildbot.openwrt.org/openwrt-19.07/packages/one_line_per_build

Likely the dependencies will straighten up soon in the download repo.