openwrt/luci #2889: luci-app-firewall

steelrat2000, 2019-07-20:
Hello. New firewall scripts do not work correctly. Bad entries in the 'firewall' file are examples. '
Good entries:
`config rule
        option target 'ACCEPT'
        option src 'wan'
        option proto 'tcp udp'
        option dest_port '21322'
        option name 'name'

config redirect
        option target 'DNAT'
        option src 'wan'
        option dest 'lan'
        option proto 'tcp udp'
        option dest_ip '10.0.0.127'
        option name 'name'
        option src_dport '1081'
        option dest_port '1081'
`
Bad entries (generated by "luci-app-firewall"):
`
config rule
        option dest_port '21322'
        list src 'wan'
        option name 'name'
        option target 'ACCEPT'
        option proto 'tcp udp'

config redirect
        option dest_port '1081'
        option proto 'tcp udp'
        option name 'nas'
        option src_dport '1081'
        list dest 'lan'
        list src 'wan'
        option dest_ip '10.0.0.2'
`
The entry "config redirect" is missing and instead of "option" is "list".

That's why the firewall is not working properly.
Steel Rat.

steelrat2000, 2019-07-21:
Its wok now. Thanks.
But a new problem appeared. When adding rules, there is no "Device" selection in the "Destination zone" field. When I develop the "Source zone", the "Device (Input)" option appears. When editing, it's OK.
Steel Rat.

jow-, 2019-07-22:
Should be fixed with the latest commits.

closed 2019-07-22 by jow-