openwrt/luci #2894: luci_statistics: bad argument #1 to 'pairs' (table expected, got nil)

fantom-x, 2019-07-20:
This happens on both R7800 & gl-ar7500 and it was all working until a day or two ago

19.07-SNAPSHOT

There is no maintainer in the Makefile, so not sure who to tag here. 

```
Failed to execute call dispatcher target for entry '/admin/statistics/graph/cpufreq'.
The called action terminated with an exception:
/usr/lib/lua/luci/statistics/datatree.lua:159: bad argument #1 to 'pairs' (table expected, got nil)
stack traceback:
	[C]: in function 'pairs'
	/usr/lib/lua/luci/statistics/datatree.lua:159: in function 'plugin_instances'
	.../luci/controller/luci_statistics/luci_statistics.lua:187: in function <.../luci/controller/luci_statistics/luci_statistics.lua:142>
	(tail call): ?

```

hnyman, 2019-07-20:
Does it happen also with master?
(I am not at home right now, so I can't check with my R7800)

It may have something to do with the recent kernel bumps, (or alternatively with the collectd config reorganisation in luci_statistics, but that happened only in master, I think. There has been no relevant change to LuCI 19.07 recently.)

fantom-x, 2019-07-20:
I cannot try master any time soon, but this was working on both routers just a few days ago in 19.07.

fantom-x, 2019-07-20:
BTW, it is not only about "cpufreq": all other plugins are throwing the same error:

```
Failed to execute call dispatcher target for entry '/admin/statistics/graph/thermal'.
The called action terminated with an exception:
/usr/lib/lua/luci/statistics/datatree.lua:159: bad argument #1 to 'pairs' (table expected, got nil)
stack traceback:
	[C]: in function 'pairs'
	/usr/lib/lua/luci/statistics/datatree.lua:159: in function 'plugin_instances'
	.../luci/controller/luci_statistics/luci_statistics.lua:187: in function <.../luci/controller/luci_statistics/luci_statistics.lua:142>
	(tail call): ?
```

fantom-x, 2019-07-20:
I moved my "rrd" directory to a USB flash drive and it happened to get mounted with 700. "chmod go+rx" fixed the problem. Turned out, parts of luci statistics are not running as root.

closed 2019-07-20 by fantom-x

jow-, 2019-07-21:
We still should improve the error handling here. Instead of `bad argument to pairs` it should probably thorw some `cannot read directory` or `cannot open file error`.

hnyman, 2019-07-21:
The advice about not running as root and the need for relaxed permissions is mentioned on the stats directory config pagr on Luci...

Good that you got it solved.

fantom-x, 2019-07-21:
Making an error message more user friendly is needed, as it Was not too trivial to figure out. An existing comment on a page is nice, but it obviously does not stop this issue from happening.

fantom-x, 2019-07-21:
Oh, can the page with the comment about needing relaxed permissions actually check the access and display a big warning if access is blocked? And also a button to fix the permissions, maybe?