openwrt/luci #2912: change network lan ip cannot be apply

ptpt52, 2019-07-23:
change lan ip then save && apply
after timeout to 0s
no force apply windows pop up
and cannot save config

`Waiting for configuration to get applied… 0s`

tested on latest HEAD code and really unusable

ptpt52, 2019-07-23:
@jow- 
it seems to be something related to your recently commits (don't know which one)

riverscn, 2019-07-23:
I met this bug, too.

ptpt52, 2019-07-23:
```
commit 843031f8197aa2c86585360c0f6d8a3b1ee02dee
Author: Jo-Philipp Wich <jo@mein.io>
Date:   Sun Jan 6 17:08:37 2019 +0100

    luci-base: luci.js: add HTTP request functions
    
    Add a fetch() inspired HTTP request utility class to luci.js and
    replace the old xhr.js class with a stub using the new request api.
    
    Signed-off-by: Jo-Philipp Wich <jo@mein.io>
```

This related?

jow-, 2019-07-23:
Check the browser debug console for errors.

ptpt52, 2019-07-23:
console log
```
luci.js?v=git-19.200.40594-b76e583:58 Uncaught (in promise) Error: XHR request aborted by browser
    at Object.handleReadyStateChange (luci.js?v=git-19.200.40594-b76e583:58)
```

ptpt52, 2019-07-23:
at this line
```
catch(e){rejectFn.call(opt,e);}});},handleReadyStateChange:function(resolveFn,rejectFn,ev){var xhr=this.xhr;if(xhr.readyState!==4)
return;if(xhr.status===0&&xhr.statusText===''){rejectFn.call(this,new Error('XHR request aborted by browser'));}
```

ptpt52, 2019-07-23:
somethings
```
luci.js?v=git-19.203.58942-a0254cb:57 GET http://192.168.56.254/ubus/?1563900586454 400 (Bad Request)
```

ptpt52, 2019-07-23:
I don't really understand the javascript...

ptpt52, 2019-07-25:
```
        handleReadyStateChange: function(resolveFn, rejectFn, ev) {
            var xhr = this.xhr;

            if (xhr.readyState !== 4)
                return;

            if (xhr.status === 0 && xhr.statusText === '') {
                rejectFn.call(this, new Error('XHR request aborted by browser'));
            }
            else {
                var response = new Response(
                    xhr, xhr.responseURL || this.url, Date.now() - this.start);

                Promise.all(Request.interceptors.map(function(fn) { return fn(response) }))
                    .then(resolveFn.bind(this, response))
                    .catch(rejectFn.bind(this));
            }
        },
```

@jow- 
hi could you help to resolve this?
it rase Error on `rejectFn.call(this, new Error('XHR request aborted by browser'));`

I don't know what it is..

ptpt52, 2019-07-25:
@jow- 
you could reproduce this error by changing lan ip from `192.168.0.1` to `192.168.100.1` when you connect to lan from PC

dibdot, 2019-07-26:
I can reproduce this behaviour while changing the lan ip address. It happens 10 seconds after the timeout started - reproducible with latest firefox and chrome on debian 10.

[image]

jow-, 2019-07-26:
Please try this change:
```diff
diff --git a/modules/luci-base/htdocs/luci-static/resources/ui.js b/modules/luci-base/htdocs/luci-static/resources/ui.js
index 1a98dbcc9..43afc698f 100644
--- a/modules/luci-base/htdocs/luci-static/resources/ui.js
+++ b/modules/luci-base/htdocs/luci-static/resources/ui.js
@@ -1929,7 +1929,7 @@ return L.Class.extend({
                                                method: 'post',
                                                timeout: L.env.apply_timeout * 1000,
                                                query: L.ui.changes.confirm_auth
-                                       }).then(call);
+                                       }).then(call, call);
                                }, delay);
                        };
```

dibdot, 2019-07-26:
The proposed change fixes the problem for me - thank you!

commit b4129640 (2019-07-26): luci-base: keep polling confirmation endpoint on XHR failures

Fixes: #2912
closed 2019-07-26 by jow-