openwrt/luci #2922: [luci-app-vpn-policy-routing] columns are too wide | wasting space and distorting the layout

n8v8R, 2019-07-26:
as described https://github.com/stangri/openwrt_packages/issues/69

n8v8R, 2019-07-26:
For some reason the outer width of the table cells in question compute to 320

[image]

dibdot, 2019-07-27:
There is no such app in this repo, please clarify your issues with the developer directly. If you're are not satisfied with the measurements of the default templates, you can change nearly every aspect of your app design with a bit of customization of these.

closed 2019-07-27 by dibdot

n8v8R, 2019-07-27:
the app developer cited

> AFAIK, this is auto-set by Luci, I have no control over this.

n8v8R, 2019-07-27:
I would appreciate this being reopened. Reason is that the ccs width is set to auto and thus not the app developer's fault that that cell is wasting space and since it happens in other LuCI modules as well

[image]

jow-, 2019-07-27:
Note that you're using an unsupported template. When checking with the default luci-theme-bootstrap, the application looks normal here.

n8v8R, 2019-07-27:
Thanks for the feedback. Indeed, out of 4 themes (bootstrap| rosy | OpenWrt | material) on the material exhibits the issue,

stangri, 2019-07-28:
@dibdot @jow : if you can educate me how can I explicitly set the widths of columns, ideally without packaging my own copy of view/cbi/tblsection, I'd be much obliged.