openwrt/luci #2924: collectd-mod-cpufreq: displays nil's in the legend

fantom-x, 2019-07-26:
OpernWRT-19.07 / R7800

See the bottom left corner of each chart. I have not yet figured out what it is supposed to be displaying there, but likely not nil's. I am running the router with the performance governor, so that could be the trigger.

[image]

@hnyman

hnyman, 2019-07-27:
As far as i remember, both cores are supposed to be on the same graph. (Can't check now)

Are you running with the default kernel and CPU topology, or have you changed processor definitions somehow?  (Related to the frequency/clock discussion in forum).

My guess is that instead of a single CPU with two cores, you may have now defined two CPUs with one core each...

fantom-x, 2019-07-27:
I also thought both used to be on the same chart. I am currently running a default build **WITHOUT** any L2 cache changes. I have stopped **collectd**, removed **cpu-0**, **cpu-1**, **cpufreq-0**, and **cpufreq-1** folders as well as **/tmp/rrdimg** and got this after starting up **collectd**.

[image]

hnyman, 2019-07-28:
@fantom-x 
You are right, the charts display now wrongly in 19.07 and master.

The reason is the upstream commit https://github.com/collectd/collectd/commit/e4896436682865ad3a1baaadb1a0ac8f3abe8fb8#diff-614f18f527573074470e543a663b90ed (and later fixes) in the collectd 5.9.0 branch that increased the amount of data collected by the cpufreq plugin. That has led to change in the data directory content (see below my R7800 with 18.06 vs 19.07). Instead of two data files in one directory, there are now directories per core, and each dir contains also additional data (total amount of freq transitions, percentage of time with each frequency).

I will need to adjust the plugin for the new data structure.

``` 
 OpenWrt 18.06-SNAPSHOT, r7820-65a405382b
 -----------------------------------------------------
root@router1:~# ls -l /tmp/rrd/router1/cpufreq/
-rw-r--r--    1 root     root          5544 Jul 28 20:11 cpufreq-0.rrd
-rw-r--r--    1 root     root          5544 Jul 28 20:11 cpufreq-1.rrd
root@router1:~#

 OpenWrt 19.07-SNAPSHOT, r10274-8efccf76f8
 -----------------------------------------------------
root@router1:~# ls -l /tmp/rrd/router1/cpufreq-*
/tmp/rrd/router1/cpufreq-0:
-rw-r--r--    1 root     root          5544 Jul 28 20:16 cpufreq.rrd
-rw-r--r--    1 root     root          5544 Jul 28 20:16 percent-1000000.rrd
-rw-r--r--    1 root     root          5544 Jul 28 20:16 percent-1400000.rrd
-rw-r--r--    1 root     root          5544 Jul 28 20:16 percent-1725000.rrd
-rw-r--r--    1 root     root          5544 Jul 28 20:16 percent-384000.rrd
-rw-r--r--    1 root     root          5544 Jul 28 20:16 percent-600000.rrd
-rw-r--r--    1 root     root          5544 Jul 28 20:16 percent-800000.rrd
-rw-r--r--    1 root     root          5544 Jul 28 20:16 transitions.rrd

/tmp/rrd/router1/cpufreq-1:
-rw-r--r--    1 root     root          5544 Jul 28 20:16 cpufreq.rrd
-rw-r--r--    1 root     root          5544 Jul 28 20:16 percent-1000000.rrd
-rw-r--r--    1 root     root          5544 Jul 28 20:16 percent-1400000.rrd
-rw-r--r--    1 root     root          5544 Jul 28 20:16 percent-1725000.rrd
-rw-r--r--    1 root     root          5544 Jul 28 20:16 percent-384000.rrd
-rw-r--r--    1 root     root          5544 Jul 28 20:16 percent-600000.rrd
-rw-r--r--    1 root     root          5544 Jul 28 20:16 percent-800000.rrd
-rw-r--r--    1 root     root          5544 Jul 28 20:16 transitions.rrd
```

fantom-x, 2019-07-28:
I hope the additional data will find its way to the charts.

hnyman, 2019-07-29:
> I hope the additional data will find its way to the charts.

Not quite sure yet. Below is example.
The value from the transitions per second is questionable, but the frequency distribution at each speed might be interesting.

I have not yet figured out if it will be possible to display the frequencies from both cores in the same chart. That would be nice. But I fear that it is not possible.

Below is example of charts that would be done separately for all cores. I think that e.g. on quad-core systems the amount of charts would be overwhelming. Might be sensible to continue showing only the main CPU frequency charts.

I have also not figured out if the frequencies in the distribution chart can be sorted. They can't be weighted in advance, as the available frequencies for different CPUs are different, so they can't be listed in advance.

[image]

Test changes done on live system:

```
root@router1:/tmp/rrd/router1# diff -u /rom/usr/lib/lua/luci/statistics/rrdtool/definitions/cpufreq.lua  /usr/lib/lua/luci/statistics/rrdtool/definitions/cpufreq.lua
--- /rom/usr/lib/lua/luci/statistics/rrdtool/definitions/cpufreq.lua    2019-07-29 21:01:08.000000000 +0300
+++ /usr/lib/lua/luci/statistics/rrdtool/definitions/cpufreq.lua        2019-07-29 23:39:26.000000000 +0300
@@ -8,22 +8,45 @@

 function rrdargs( graph, plugin, plugin_instance, dtype )

-       return {
-               title = "%H: Processor frequency",
+       local cpufreq =  {
+               title = "%H: Processor frequency - core %pi",
                alt_autoscale = true,
                vlabel = "Frequency (Hz)",
                number_format = "%3.2lf%s",
                data = {
-                       sources = {
-                               cpufreq = { "" }
-                       },
+                       types = {"cpufreq" },
                        options = {
-                               cpufreq_0 = { color = "ff0000", title = "Core 0", noarea=true, overlay=true },
-                               cpufreq_1 = { color = "0000ff", title = "Core 1", noarea=true, overlay=true },
-                               cpufreq_2 = { color = "00ff00", title = "Core 2", noarea=true, overlay=true },
-                               cpufreq_3 = { color = "00ffff", title = "Core 3", noarea=true, overlay=true }
+                               cpufreq = { color = "ff0000", title = "Frequency", noarea=true, overlay=true },
                        }
                }
        }
+
+       local transitions = {
+               title = "%H: Frequency transitions - core %pi",
+               alt_autoscale = true,
+               vlabel = "Transitions per second",
+               number_format = "%3.2lf%s",
+               data = {
+                       types = { "transitions" },
+                       options = {
+                               transitions = { color = "0000ff", title = "Transitions", noarea=true, overlay=true },
+                       }
+               }
+       }
+
+       local percentage = {
+               title = "%H: Frequency distribution - core %pi",
+               alt_autoscale = true,
+               vlabel = "Frequency (Hz)",
+               number_format = "%3.2lf%s",
+               data = {
+                       types = { "percent" },
+                       options = {
+                               percent = { title = "Frequency %di" },
+                       }
+               }
+       }
+
+       return { cpufreq, transitions , percentage}
 end

```

hnyman, 2019-07-30:
Fixed with e6f30bb7 in master and with dd908ced in 19.07.

I enabled the new data, but decided to hide it behind a new option "ExtraItems" by default, as otherwise the total amount of per-core graphs could grow quite much.

I consider the information value of the transition counts to be low, but the value of the longer-term data on frequency distribution might be interesting, especially if the frequencies could be sorted. The current random order make the chart less valuable.

As the frequencies vary by CPU, no static predefined "weights" can be defined in the app, making the sorting currently random. 

I have considered two approaches for sorting: 

* sorting by data instance name. Works. I modified locally the weight-based sorting functionality (that was introduced by @jow- a few years ago) by adding the instance name (frequency) as a new fallback alternative as a sorting factor:
  ```
  local x = a.weight or a.index or 0
  -->
  local x = a.weight or instance or a.index or 0
  ```
  That worked, but as the instance name is handled as string, sorting is alphabetical (like 100, 120, 60, 80), which is not what is targeted.
  I also consider that a bit invasive, as I have not checked all other stats graphs for possible negative fallout.

* The other approach that I thought, is defining a new option that would say to "use the numerical value of the data instance name as the weight". Then the sorting function itself could stay unmodified, and there would just be a new option that would be set the weight.

closed 2019-07-30 by hnyman

hnyman, 2019-07-31:
I enhanced the functionality further.with 4ca5727 and utilised that in 05cad42

A numeric data instance name can now be used for sorting, and coloring can be done using pre-defined colors after the sorting.

This enables nice CPU frequency usage charts:
[image]

fantom-x, 2019-07-31:
This is very cool! I will give it a shot once the packages are refreshed.