openwrt/luci #2926: Linksys WRT1900ACS "Unsupported device, release, target, subtraget or board"

apcimino, 2019-07-27:
I can request a new firmware but I get the "unsupported device..." error message.

Please let me know if I can provide any more details.

New release 18.06.4 available
Installed version: 18.06.1
Package upgrades available
luci-lib-nixio: git-19.188.21194-7d8df17-1 to git-19.181.61051-9bab4ba-1
luci-lib-jsonc: git-19.188.21194-7d8df17-1 to git-19.181.61051-9bab4ba-1
luci-theme-bootstrap: git-19.188.21194-7d8df17-1 to git-19.181.61051-9bab4ba-1
luci-app-wifischedule: git-19.188.21194-7d8df17-1 to git-19.181.61051-9bab4ba-1
luci-app-adblock: git-19.188.21194-7d8df17-1 to git-19.181.61051-9bab4ba-1
dnsmasq: 2.80test3-1 to 2.80-1.4
luci-lib-ip: git-19.188.21194-7d8df17-1 to git-19.181.61051-9bab4ba-1
luci-base: git-19.188.21194-7d8df17-1 to git-19.181.61051-9bab4ba-1
luci-app-attendedsysupgrade: git-19.188.21194-7d8df17-1 to git-19.181.61051-9bab4ba-1
luci-app-firewall: git-19.188.21194-7d8df17-1 to git-19.181.61051-9bab4ba-1
luci-proto-ipv6: git-19.188.21194-7d8df17-1 to git-19.181.61051-9bab4ba-1
luci-proto-ppp: git-19.188.21194-7d8df17-1 to git-19.181.61051-9bab4ba-1
luci: git-19.188.21194-7d8df17-1 to git-19.181.61051-9bab4ba-1
luci-mod-admin-full: git-19.188.21194-7d8df17-1 to git-19.181.61051-9bab4ba-1

dibdot, 2019-07-27:
Maybe a temp. server issue!? @aparcar

adelton, 2025-01-10:
@apcimino Was it a one time issue that could have been attributed to a temporary server problem, or is requesting a new problem still broken on your device?

apcimino, 2025-01-10:
My guess, 5 years later, is this was probably a temp server issue.
I don’t have that router anymore so wouldn’t be able to test again.

systemcrash, 2025-01-10:
transient

closed 2025-01-10 by systemcrash