openwrt/luci #2927: [bug] luci-theme-material table cell are too wide | wasting space and distorting the layout

n8v8R, 2019-07-27:
as described in https://github.com/openwrt/luci/issues/2922 `luci-theme-material` (git-19.207.37184-d89a027-1.0) is displaying/computing table cells too wide and thereby wasting space and distorting the layout

jow-, 2019-07-27:
@LuttyYang

n8v8R, 2019-08-14:
@jow is the source code for `luci-theme-material` even being still actively maintained or development abandoned and just sporadically patched in downstream [OpenWrt](https://github.com/openwrt/luci/tree/master/themes/luci-theme-material) , considering that the last [upstream commit](https://github.com/LuttyYang/luci-theme-material) is dating back to 21 Sep 2017?

adelton, 2025-01-10:
Just for the record, I compared Bootstrap and Material and the widths in the Logging table (as shown in https://github.com/openwrt/luci/issues/2922#issuecomment-515668470) are virtually the same:

[image]
[image]

I understand that the themes might have changed since 2019 -- is there still anything of concern in 2025?

closed 2025-01-10 by systemcrash