openwrt/luci #2937: luci on lighttpd- 5000 internal server error

syedumerbukhari, 2019-07-30:
any solution?

Lua 5.1.5  Copyright (C) 1994-2012 Lua.org, PUC-Rio
> require "luci.cacheloader"
> require "luci.sgi.cgi"
> 
> luci.dispatcher.indexcache = "/tmp/luci-indexcache"
> 
> luci.sgi.cgi.run()
.../luci/host//usr/lib/lua/luci/statistics/datatree.lua:28: attempt to index upvalue 'sections' (a nil value)
stack traceback:
	.../luci/host//usr/lib/lua/luci/statistics/datatree.lua:28: in function '__init__'
	/home/luci/host//usr/lib/lua/luci/util.lua:70: in function 'Instance'
	.../luci/controller/luci_statistics/luci_statistics.lua:30: in function 'v'
	/home/luci/host//usr/lib/lua/luci/dispatcher.lua:496: in function 'createtree'
	/home/luci/host//usr/lib/lua/luci/dispatcher.lua:179: in function 'dispatch'
	/home/luci/host//usr/lib/lua/luci/dispatcher.lua:143: in function </home/luci/host//usr/lib/lua/luci/dispatcher.lua:142>
Status: 500 Internal Server Error
Content-Type: text/plain
Cache-Control: no-cache
Expires: 0

jow-, 2019-07-30:
Loooks like a permission issue. Ensure that the LuCI CGI process runs as root.

closed 2019-07-30 by jow-

syedumerbukhari, 2019-07-30:
thanks, this one solved now I'm missing some libraries. on web page 500 internal server error
and when I checked lighttp server-status I got this message of missing files.
I can't find these files.

7月 30 14:59:07 toor-VirtualBox lighttpd[20600]:         no file '/usr/local/lib/lua/5.1/loadall.so'
7月 30 14:59:07 toor-VirtualBox lighttpd[20600]:         no file './luci.so'
7月 30 14:59:07 toor-VirtualBox lighttpd[20600]:         no file '/usr/local/lib/lua/5.1/luci.so'
7月 30 14:59:07 toor-VirtualBox lighttpd[20600]:         no file '/usr/lib/x86_64-linux-gnu/lua/5.1/luci.so'
7月 30 14:59:07 toor-VirtualBox lighttpd[20600]:         no file '/usr/lib/lua/5.1/luci.so'
7月 30 14:59:07 toor-VirtualBox lighttpd[20600]:         no file '/usr/local/lib/lua/5.1/loadall.so'
7月 30 14:59:07 toor-VirtualBox lighttpd[20600]: stack traceback:
7月 30 14:59:07 toor-VirtualBox lighttpd[20600]:         [C]: in function 'require'
7月 30 14:59:07 toor-VirtualBox lighttpd[20600]:         /var/www/html/cgi-bin/luci:2: in main chunk
7月 30 14:59:07 toor-VirtualBox lighttpd[20600]:         [C]: ?