openwrt/luci #2938: luci-base: list element only get saved if + button is explicit pressed before save&apply

feckert, 2019-07-30:
- Perceived behavior:
If you do not press explicit the **"+"** button on a list element before **save&apply** the value will not get saved in the configuration and is lost. 

- Expected behavior:
If I add a new element to a list value and do **not** press explicit the **"+"** button before pressing **save&apply** the configuration should be saved anyway.

- Procedure for reproduction:
On the LuCI page go to **/cgi-bin/luci/admin/system/system** and add a new time sync server. without explicit press the **"+"** button before **save&apply**

I do not know since this has changed! But this has worked definitely before. I think this has something todo with javascript but @jow- I am helpless what the problem could be.

feckert, 2019-07-30:
Thanks @jow- 
That was really quick.
:+1:

feckert, 2019-07-30:
This also does not work for openwrt-19.06.
There were a lot of changes!
So could you please also fix this for this branch as well?

jow-, 2019-08-14:
@feckert - done.

feckert, 2019-08-16:
@jow-  on **openwrt-19.07** and **master** it is working now again as expected thanks :+1: for your work

feckert, 2019-08-19:
@jow- Apparently, it fit. But when looking more closely it only works when the button **Save** is pressed. When **Save & Apply** is pressed the value is **not** taken over. Sorry for the bad news :-(

feckert, 2019-09-16:
Thanks now it works as expected :-)

commit 3880c3f2 (2019-07-30): luci-base: ui.js: save filled out but not added dynlist items on submit

Fixes: #2938
commit 0fc125a8 (2019-08-14): luci-base: cbi.js: save filled out but not added dynlist items on submit

Fixes: #2938
commit 8a2f1c76 (2019-09-13): luci-base: cbi.js: save filled out but not added dynlist items on apply

Fixes: #2938
closed 2019-09-16 by feckert