openwrt/luci #2940: Luci returns Blank Page

syedumerbukhari, 2019-08-01:
hi, I'm having trouble while running LUCI interface on Lighttpd server.

when I access the URL LUCI returns a black page with a link: "LuCI - Lua Configuration Interface"
and it downloads LUCI file when I click on this URL.

i have edited /etc/lighttpd/lighttpd.conf file according to the guide given on official site.
after editing when I accessed the web interface it just redirects me to IP/cgi-bin/luci with blank page.

any suggestion :(

jow-, 2019-08-01:
The logread for errors. Note that LuCI on Lighttpd is unsupported, consider running it under uhttpd instead and use lighthttpd for reverse proxying.

closed 2019-08-01 by jow-

syedumerbukhari, 2019-08-01:
brother thanks for your reply :)
I'm confused that on the official website it's written we can run it on many servers like apache etc.

Actually, I'm trying to run LUCI-V0.9 on ubuntu-16.04 fro development later I'm planning to move it on my router which is running ARM kernel 4.1.15.

in the terminal when I run systemctl status Lighttpd it gives me this error:

8? 01 16:20:55 toot-VirtualBox lighttpd[19095]:         no file '/usr/local/lib/lua/5.1/loadall.so'
8? 01 16:20:55 toot-VirtualBox lighttpd[19095]:         no file './luci.so'
8? 01 16:20:55 toot-VirtualBox lighttpd[19095]:         no file '/usr/local/lib/lua/5.1/luci.so'
8? 01 16:20:55 toot-VirtualBox lighttpd[19095]:         no file '/usr/lib/x86_64-linux-gnu/lua/5.1/luci.so'
8? 01 16:20:55 toot-VirtualBox lighttpd[19095]:         no file '/usr/lib/lua/5.1/luci.so'
8? 01 16:20:55 toot-VirtualBox lighttpd[19095]:         no file '/usr/local/lib/lua/5.1/loadall.so'
8? 01 16:20:55 toot-VirtualBox lighttpd[19095]: stack traceback:
8? 01 16:20:55 toot-VirtualBox lighttpd[19095]:         [C]: in function 'require'
8? 01 16:20:55 toot-VirtualBox lighttpd[19095]:         /var/www/html/cgi-bin/luci:2: in main chunk
8? 01 16:20:55 toot-VirtualBox lighttpd[19095]:         [C]: ?

jow-, 2019-08-01:
Seems the Lua CGI executable is not able to find its libraries. Note that running LuCI outside of OpenWrt hasn't been supported for years, so you'll be better off using an OpenWrt x86 virtual machine for development.

syedumerbukhari, 2019-08-01:
got it, brother.
Thanks.
I will try it on QEMU-KVM :)