openwrt/luci #2944: [Bug] Interface name length validation not performed

Selora, 2019-08-01:
Using 18.06

In the interface creation menu (as per screenshot), if the interface name length is close to 15 characters (the hard maximum length for a Linux interface), and the interface is bridged, the menu allows the user to create the interface.

However, OpenWRT (Luci?) will add the "br-" prefix to the interface name, thus making it longer than 15 characters. The bridge is never created and there's no log I can tell that explicitly mention why the bridge creation failed.

Proposed solution: 
* Easy way: allow names no longer than 12 characters 
* Hard way: Prevent bridging iface with names > 12 characters OR remove the "br-" prefix.

For reference, I think just modifying this would work:

 https://github.com/openwrt/luci/blob/0f91ef83afa5236c866605515994a63fa363a65b/modules/luci-mod-network/luasrc/model/cbi/admin_network/iface_add.lua#L24

hnyman, 2019-08-01:
Not that easy, because the prefix may also be "6in4-", "6to4-", "pppoe-" etc., or there may be no prefix.

See earlier references.

* Reference to bug report & discussion from year 2015 in the old OpenWrt bug tracker:
https://dev.archive.openwrt.org/ticket/20380.html
* https://github.com/openwrt/luci/issues/507
* https://github.com/openwrt/luci/issues/345
* https://github.com/openwrt/luci/issues/2740

comments also in commit
* https://github.com/openwrt/luci/commit/b1217c88c3566c1bd726bce9203da591af564bcf

Selora, 2019-08-01:
I see, it was written as plain as day right below where I cropped in the screenshot.
Yet I managed to not read it and spend hours trying to figure out what was wrong.
The fault is 100% on me, I was looking everywhere else than this page.

I see that it's not a small fix. May I suggest adding some client-side verification for length, and when the user start typing, highlighting and/or popup this warning?

It's just something that I never encountered after months of using OpenWRT, and since I'm becoming familiar with the menu I was paying very little attention to the text in the menu itself.

Again, thanks for all the work! I know you probably have thousands of more important things to do than foolproof the interface for people like me that don't even bother reading!
Cheers.

feckert, 2019-09-13:
On master I see a red hint
**Expecting: Value with at most 15 characters**
So I think we can close this.

closed 2019-09-13 by feckert