openwrt/luci #2973: Luci interface boxes are off-center

lorddoskias, 2019-08-09:
I just upgraded to openwrt 18.06.4 and observed that some elements on the main status page are off-center. The exact version of luci (according to openwrt ) is: **Powered by LuCI openwrt-18.06 branch (git-19.170.32094-4d6d8bc)**

On the [attached picture](https://ibb.co/kJ0GWGp) you can clearly see that: 

 * The box containing device right below IPv4/6 upstream is placed awkardly. It's not clear where it should be in the upstream boxes or below it. 
  * Boxes describing the wireless interfaces are also placed awkwardly and cross the border of the 2 boxes, describing radio0/radio1

My browser is chromium from ubuntu 18.04

feckert, 2019-08-09:
What theme do you use

lorddoskias, 2019-08-09:
I'm using whatever is the default with openwrt. If I navigate to 'System'->'Language and Style' I see 'design' is set to bootstrap. And when looking at installed packages I only have: 

luci-theme-bootstrap	git-19.170.32094-4d6d8bc-1

lorddoskias, 2019-08-09:
I just upgraded to: git-19.218.48077-4468345-1	and the issue seems fixed.

closed 2019-08-09 by lorddoskias