openwrt/luci #2985: luci-app-upnp: Cannot see added IPv6 UPnP/PCP rules in LuCI interface

ffrediani, 2019-08-16:
When dynamically adding a IPv6 Firewall rule (AddPinhole) making a request to miniupnpd 2.1 in OpenWrt 18.06 it adds the rule correctly to ip6tables chain and allow incoming traffic towards the target device in the LAN.

However on luci-app-upnp it doesn't show anything in the UPnP/PCP LuCI interface as it does for IPv4 added rules. This may be a problem for users that may not have familiarity to command line if a added rule must be removed before the lease time expires or when there is no lease time to avoid security issues.

luci-app-upnp version used is git-19.218.48077-4468345-1

feckert, 2019-08-16:
The luci-app-upnp supports for now only Ipv4.
Feel free to extend luci-app-upnp to support also ipv6

ffrediani, 2019-08-16:
Hello. I think it is clear that luci-app-upnp only supports IPv4.
I didn't come to propose I will extend support to IPv6 but to ask that to those who did for IPv4 or have the skills to do for IPv6 and bring proper justifications for that.

In 2019 and that amount of IPv6 availability in fixed and mobile broardband around the world it seems reasonable to have IPv6 properly supported in all its aspects in OpenWrt, which by the way has been of the key pushers for IPv6 deployment in many scenarios where commercial CPEs lack proper IPv6 support at the current times.