openwrt/luci #2986: [bug] mount points section showing wrong device as mounted to root

n8v8R, 2019-08-16:
* 19.07
* luci-mod-admin-full git-19.227.85747-c9d8a9b-1.0
* busybox 1.30.1-5.2
____

From IRC understood that LuCI is parsing output from `df` (busybox implemented). And there is a discrepancy to the output of `mount`

Upon device boot

* `mount` produces 
> **/dev/mmcblk0p1** on /

whilst 

* `df`

> Filesystem           Mounted on
> /dev/root  /

[image]
____
now cli exec `mount /dev/mmcblk0p1 /` which prints

> mount: /: /dev/mmcblk0p1 already mounted on /.

and thereafter LuCI showing (correctly)

[image]

n8v8R, 2019-08-16:
Whilst this blog post https://bootlin.com/blog/tag/busybox/ mentions `rdev` it does not seem implemented with `busybox` or otherwise available in OpenWrt

dangowrt, 2019-08-16:
It is implemented in `block info` which does resolve the actual block device by getting major and minor device id via rdev.
see https://git.openwrt.org/?p=project/fstools.git;a=blob;f=block.c;h=39212d247fe46e07b4ea4113edf4687b59b7d4e7;hb=HEAD#l656

n8v8R, 2019-08-16:
Alright, thanks for the pointer.

It is getting stranger 

device rebooted and `block info` printing

> label buffer too small 256 > 63
> label buffer too small 256 > 63
> /dev/mmcblk0p1: UUID="75b9e7e0-d26e-442b-bc5a-7482c8b82b54" TYPE="btrfs"

-> notice lack of 

> MOUNT="/srv"

and after cli exec `mount /dev/mmcblk0p1 /` `block info` printing

> label buffer too small 256 > 63
> label buffer too small 256 > 63
> /dev/mmcblk0p1: UUID="75b9e7e0-d26e-442b-bc5a-7482c8b82b54" **MOUNT="/"** TYPE="btrfs"
____
That makes it now **three different results**

* `mount` - saying that /dev/mmcblk0p1 is mounted on /
* `df` - saying that /dev/root is mounted on /
* `block info` - nothing about /dev/root but /dev/mmcblk0p1 not being mounted at / (all)

n8v8R, 2019-08-17:
The (underlying) cause seems to have been introduced with code changes between 06 Aug and 12 Aug, during which period the kernel been bumped from 4.14.134 to .13**5** / .13**6** / .13**7**. Not sure what other patches (general or particular to target mvebu) may have been committed that could this

n8v8R, 2019-08-17:
the issue seems to be caused by `btrfs device scan` not working since being moved to preinit https://github.com/openwrt/packages/commit/ae48f3f9e90ccae04a4b9e1946594cf62f217f4a

manually cli exec `btrfs device scan` after boot resolves the issue, with `mount` & `df` & `block info` being in alignment/agreement about the device mounted to /

closed 2019-08-17 by n8v8R