openwrt/luci #2997: [BUG] Unbridge Interface leads to bad "ifname" value

sch-m, 2019-08-22:
@jow- 
When you Uncheck the "Bridge Interfaces" Feature of a bridge interface with more than one selected physical interfaces, the ifname value is not properly cleaned up after save&apply.

I think this is related to this call to get_interfaces() in the ifname_single.write function
https://github.com/openwrt/luci/blob/2aab85814d0eead5f82ee1c5661f03667ce50b62/modules/luci-mod-network/luasrc/model/cbi/admin_network/ifaces.lua#L308
because get_interfaces checks for self:is_bridge(), which is not the case after unchecking the "Bridge Inferfaces" checkbox. Therefore, the old_ifs[] array/list does not include all the "old" interfaces.

BTW: Why are the old_ifs explicitly removed here. And what is this backup_ifnames() good for? :-) Can't we just flush the ifname option and add the new selected interface(s)?

jow-, 2019-08-22:
> Can't we just flush the ifname option and add the new selected interface(s)?

No because the `get_interfaces()` list might also include special wireless identifiers whose removal must be delegated to the wireless config. Merely flushing `option ifname` will not remove the association of `wifi-iface` entries to the bridge being unbridged.

sch-m, 2019-08-22:
Ah OK, good point. So next question: why is get_interfaces() checking for is_bridge()? Is this really necessary?

jow-, 2019-08-22:
Not sure of the implications of removing the bridge type check right now. A quick fix would be adding some kind of `ignore_bridge_state` argument to `get_interfaces()` which makes it skip the bridge check internally.

jow-, 2019-08-22:
Before investing too much effort into this, can you please check if https://github.com/jow-/luci works as expected? I plan to merge this to the master soon.

sch-m, 2019-08-22:
I will test it tomorrow. But we also need a fix for openwrt-19.07.

sch-m, 2019-08-23:
> Before investing too much effort into this, can you please check if https://github.com/jow-/luci works as expected? I plan to merge this to the master soon.

With the new "client side interface configuration pages" it works as expected.

jow-, 2019-08-23:
@sch-m - thanks for confirming. For openwrt-19.07 can you please try the following fix:
```diff
diff --git a/modules/luci-base/luasrc/model/network.lua b/modules/luci-base/luasrc/model/network.lua
index a36a23f32..c8b62f290 100644
--- a/modules/luci-base/luasrc/model/network.lua
+++ b/modules/luci-base/luasrc/model/network.lua
@@ -1255,8 +1255,8 @@ function protocol.get_interface(self)
        end
 end
 
-function protocol.get_interfaces(self)
-       if self:is_bridge() or (self:is_virtual() and not self:is_floating()) then
+function protocol.get_interfaces(self, ignore_bridge_state)
+       if ignore_bridge_state or self:is_bridge() or (self:is_virtual() and not self:is_floating()) then
                local ifaces = { }
 
                local ifn
diff --git a/modules/luci-mod-network/luasrc/model/cbi/admin_network/ifaces.lua b/modules/luci-mod-network/luasrc/model/cbi/admin_network/ifaces.lua
index de7b8676e..3893ddb38 100644
--- a/modules/luci-mod-network/luasrc/model/cbi/admin_network/ifaces.lua
+++ b/modules/luci-mod-network/luasrc/model/cbi/admin_network/ifaces.lua
@@ -305,7 +305,7 @@ if not net:is_floating() then
                if alias then
                        old_ifs[1] = '@' .. alias
                else
-                       for _, i in ipairs(net:get_interfaces() or { net:get_interface() }) do
+                       for _, i in ipairs(net:get_interfaces(true) or { net:get_interface() }) do
                                old_ifs[#old_ifs+1] = i:name()
                        end
                end

```

sch-m, 2019-08-23:
@jow- Yes, this fixes the problem for openwrt-19.07.

But I've found another problem in openwrt-19.07 related to this one:
When you uncheck the "Bridge Interfaces" checkbox of a bridge interface with proto==static, the
the interfaces dropdown is empty/damaged. You can't select anything.
With proto==ignore or proto==dhcp everything is fine.

This problem also does not exist in the latest master.

Any ideas?

jow-, 2019-08-23:
No ideas, sorry.

sch-m, 2019-08-23:
Anyone else? :-)

sch-m, 2019-08-26:
I've found out that the problem also exists in the other way: When you try to make a bridge out of an Interface with proto=="static". And it is also important that ipaddr and netmask is configured.

Really strange.

In the Firefox "inspector" it seems that in the error case there are no eventListener registered to the cbi-dropdown object.

sch-m, 2019-09-11:
Thanks for your changes. This will fix this issue.

commit fe0e4f8b (2019-09-10): luci-base, luci-mod-network: fix bridge state change quirks

Fixes: #2997
commit ff4c722a (2019-09-10): luci-base, luci-mod-network: fix bridge state change quirks

Fixes: #2997

(cherry picked from commit fe0e4f8b4d50cb7ce4e65f747d4ed2de6a030b73)
closed 2019-09-11 by sch-m