openwrt/luci #2998: [feature suggestion] luci-app-ipv6: Add NDP-Proxy option to LuCI when protocol is DHCPv6 Client

ffrediani, 2019-08-23:
Sometimes when the ISP doesn't doesn't provide a IPv6 Prefix Delegation or when it only provides a /64 to be used by the LAN interface it may be necessary to enable NDP-Proxy in order that other networks can have IPv6 connectivity.

In order to archive that the LAN interface 'Router Advertisement-Service', 'DHCPv6-Service and NDP-Proxy must be all changed to 'Relay Mode'. That can all be done fine in LuCI under DHCP Server options.
However according to OpenWrt documentation it is necessary to also add the same options (plus a fourth one - 'option master 1') to the wan interface in the /etc/config/dhcp file (https://oldwiki.archive.openwrt.org/doc/uci/network6#router_advertisement_dhcpv6).

This can only be done via CLI and cannot be done via LuCI for the WAN6 interface when the selected Protocol is 'DHCPv6 Client'.

Please add an option that allows the user to easily enable this and select which WAN interface is the master for DHCPv6, RA and NDP when in Relay mode.

This is useful because many people inadvertently connects the LAN port of a home router to a WAN port of a second router in order to extend signal and by using this technique allows that users connected to the secondary routers keep IPv6 connectivity.

feckert, 2019-09-13:
Please provide the uci network config options which have do be set by the luci

Strykar, 2020-12-28:
Seems like the checkbox option available under `LAN` for `Master`, (`Set this interface as master for the dhcpv6 relay.`)
But for the `wan6` interface when configured as a DHCPv6 client:
```
# uci show dhcp.wan6 | grep master
dhcp.wan6.master='1'
```

ailin-nemui, 2021-03-22:
the problem is that the DHCP/RA/NDP/ relay options are hidden from the UI when configured as DHCPv6 client (wan6), they are only made visible in the "Static" configuration. Thus, the fix should be rather simple -- show the "DHCP IPv6" tab also on the DHCPv6 client mode interface

ailin-nemui, 2021-03-22:
although logically it could also make sense to add a dedicated "Configure relay master to be on interface [wan6]" option to the LAN config instead

ffrediani, 2021-03-22:
Exactlly @ailin-nemui 
It's pretty much that.