openwrt/luci #2999: Missing switch menu if no switch settings

Neustradamus, 2019-08-25:
The Switch menu is missing if there is no switch settings.

Impossible to manage VLANs...

Can you resolve this bug?

I create this ticket because it is requested by @CodeFetch @adrianschmutzler: It is a LuCI problem, not OpenWrt.

cc @jow-.

CodeFetch, 2019-08-25:
@Neustradamus It's not a Luci issue in this case. Luci does not show a switch setting, because you do not have one. The problem is that your device uses the default case here https://github.com/openwrt/openwrt/blob/e1dcfe02b2af0eea1654e491f62049e479978951/target/linux/ath79/base-files/etc/board.d/02_network#L302

And there is no switch-setting created. The question is whether it makes sense to create a switch-config for two-port ath79 devices, too and what the correct way is/whether it must be decided on a per-device basis.

adrianschmutzler, 2019-08-25:
@CodeFetch I would argue against that this is to be solved on a device basis.
It is also a general question whether you want to be able to define VLANs on a one-port or two-port, e.g. when you want to connect to a network with VLANs enabled directly.

I would not call this a bug in luci, but still it is a general question how to deal with that.
If we solve this on a device-basis, it will just be anyone doing anything arbitrary, depending on his taste or use-case.

CodeFetch, 2019-08-25:
@adrianschmutzler I agree with that. My point (which is wasn't aware of, because I don't use Luci regularly) is that the Luci section is called "Switch configuration" and not "VLAN configuration", isn't it?
Maybe the correct way to define VLAN networks in Luci is to create a network in "Network configuration" e.g. with the device eth0.X where X is the VLAN ID, but maybe I'm missing something here... I'm really unsure about the correct way to handle this.

On the other hand when we have a switch already, even if it only has one port, the correct way might be to define a single-port-switch, because it could cause configuration issues to use VLANs without defining them on the switch.

Neustradamus, 2019-08-25:
"Switch" in "Network" menu

Example, in the page:
- Switch "switch0" (AR934X built-in switch)
-- Enable VLAN functionality
-- Enable mirroring of incoming packets
-- Enable mirroring of outgoing packets

- VLANs on "switch0" (AR934X built-in switch)
-- VLAN ID ...
-- Port status: ...

jow-, 2019-08-26:
LuCI relies on proper default configuration to decide whether to display a switch menu or not. If the default config of the device ships without switch setting despite there being swconfig support, then it needs to be fixed in OpenWrt.

closed 2019-08-26 by jow-

Neustradamus, 2019-08-26:
@jow-: How to manage VLANs if there is no switch?

jow-, 2019-08-26:
By specifying custom ethX.Y in the physical interface settings.

Neustradamus, 2019-08-26:
@CodeFetch @adrianschmutzler @ pep2k: Have you seen?

Neustradamus, 2019-09-02:
Any news on this missing part?

How we can create a ticket for OpenWrt?

adrianschmutzler, 2019-09-02:
I think it would be best to write to the OpenWrt mailing list and present the problem there. You never know whether you will get a response there, but I think we should at least try to get this solved in a general manner.

If there is no response and I do not have a striking inspiration until then, we can still change single or all devices to switch-based setup (which will be painful, too.).