openwrt/luci #3001: Security! Certificate problem for HTTPS of LuCI

Neustradamus, 2019-08-25:
By default no secure LuCI!
- https://fabianlee.org/2016/09/19/openwrt-enabling-https-for-the-luci-web-admin-interface/

After install of needed part, there is an automaticaly generation of the certificate!

Bad values:
- https://github.com/openwrt/openwrt/blob/master/package/network/services/uhttpd/files/uhttpd.init#L34

```
C=ZZ
ST=Somewhere
L=Unknown
O=OpenWrt+ID
CN=OpenWrt
```

Can you change it?
Add a possibility to admin to choice values?

hnyman, 2019-08-25:
???

/etc/config/uhttpd contains the config items for those data fields

And why do you claim that those values make it more unsecure? 
In any case the certificate will be a self-signed cert without proper FQDN, so most browsers will flag that cert somewhat unsecure in any case.

closed 2019-08-25 by hnyman

Neustradamus, 2019-08-25:
@hnyman: By default no HTTPS://openwrt/ only HTTP://openwrt/

After adding package in the router, uhttpd creates a certificate but the value is not good.

I have noted previously:
- https://github.com/openwrt/openwrt/blob/master/package/network/services/uhttpd/files/uhttpd.init#L34

```
C=ZZ
ST=Somewhere
L=Unknown
O=OpenWrt+ID
CN=OpenWrt
```

There is no option for change values by mines in LuCI or at generation no request to choose...

Neustradamus, 2019-08-25:
Please reopen this ticket, it is not solved.

hnyman, 2019-08-25:
Due to the aim for small footprint, no SSL support is added by default. HTTP is the default.

You can change the SSL keygen values in /etc/config/uhttpd if you like, and you can naturally re-generate the keys.  But for self-signed certs, the values do not have much value in any case.

You can also install letsencrypt support (acme) and use that if you have a full domain. 

And of course you can manually import your own certs if they are in a suitable format.

Neustradamus, 2019-08-25:
@hnyman:
- Where we can add HTTPS support in LuCI?
- Where we can select values in LuCI?
- Where we can add certs in LuCI?

LuCI is here to be easily for end users ;)

hnyman, 2019-08-25:
> Where we can enable HTTPS support in LuCI?

* system / software --> install luci-ssl

> Where we can select values of the default certificate in LuCI?

* install luci-app-uhttpd and use that to edit values. 
* or edit /etc/config/uhttpd directly

> Where we can add Lets Encrypt support in LuCI?

* install luci-app-acme

LuCI is there to simplify some of the configuration tasks, mostly the common ones that typically touch the wide userbase. But it is not an exhaustive GUI containing support for all possible config combinations.

Neustradamus, 2019-08-25:
It will be nice to add in the LuCI GUI ;)

Neustradamus, 2019-08-26:
@jow-: What do you think?

Any possibilities to add in LuCI points that I have said before?

Neustradamus, 2021-01-09:
Thanks @ynezz, it is linked to:
- https://github.com/openwrt/luci/pull/4659
- https://github.com/openwrt/luci/pull/4660

Neustradamus, 2021-09-28:
Linked to:
- https://github.com/openwrt/luci/commit/7c943a1d6bcf449019ca8a43e800e51f269bb8f6