openwrt/luci #3003: Signed overflow into Network, Interfaces : TX: 3.37 TB (-1684077764 Pkts.)

julienrobin28, 2019-08-25:
Hi, everything is in the title - as you can see :

[image]

ifconfig gives signed values for packets count, so I guess it's a luci display problem ?

jow-, 2019-08-26:
Can you please provide me with the JSON output of `http://192.168.28.1/cgi-bin/luci/admin/network/iface_status/lan` ? I want to know if its a browser-side JS formatting issue or if its already delivered wrongly by the backend.

julienrobin28, 2019-08-26:
Hi, here is the raw output :
`[{"rx_bytes":2174721365644.0,"ifname":"br-lan","tx_bytes":3435269054215.0,"ipaddrs":["192.168.28.1\/24"],"tx_packets":2658388907.0,"dnsaddrs":[],"rx_packets":1748522453,"proto":"static","is_dynamic":false,"id":"lan","subdevices":[{"rx_bytes":2221652805546.0,"type":"ethernet","tx_bytes":3435107501051.0,"typename":"Ethernet Adapter","name":"eth0","rx_packets":2094700098,"tx_packets":2658269680.0,"macaddr":"DC:A6:32:17:7E:9E","is_up":true,"ifname":"eth0"},{"rx_bytes":0,"type":"wifi","tx_bytes":0,"typename":"Wireless Adapter","name":"Master \"OpenWrt\"","rx_packets":0,"is_up":false,"tx_packets":0,"ifname":"radio0.network1"}],"ip6addrs":["fd26:b3ad:7d67::1\/60"],"desc":"Static address","is_auto":true,"typename":"Bridge","uptime":250739,"is_up":true,"type":"bridge","macaddr":"DC:A6:32:17:7E:9E","name":"br-lan"}]`

And the nicely formatted one :
[image]

commit 41e2258d (2020-01-29): luci-base: cbi.js: fix unintended number sign overflow in format

Fixes: #3003

(backported from commit 6d9a23af60f796e14dd9652d4f401db1a5ac456f)
closed 2019-09-10 by jow-