openwrt/luci #3004: LuCI backup does not follow collectd.conf symlink

thatso, 2019-08-25:
My system: Meraki MR18 with OpenWrt 18.06.4 r7808-ef686b7292 and LuCI openwrt-18.06 branch (git-19.170.32094-4d6d8bc).
I've just setup collectd and realized that backups created in LuCI don't take into account that `/etc/collectd.conf` is a symlink to `/var/etc/collectd.conf`.
Apparently, only the symlink gets backed up and not the config file itself.
My backup archive looks like this:

[image]

yousong, 2019-08-26:
In OpenWrt, `/var` by default is a symlink to `/tmp` and is not expected to be persistent across reboot.

In this specific case, `/var/etc/collectd.conf` will be recreated by from configs under `/etc/config` which is included in the backup by default.

jow-, 2019-08-26:
LuCI relies on `sysupgrade` to create the backup. Apart from that the backup is designed to back up the file system contents verbatim and only within the paths configured in either `/etc/sysupgrade.conf`, `/lib/upgrade/keep.d/*` or as provided by `opkg list-changed-conffiles`. Turning symlinks into files or copying files outside of the configured backup locations would violate this expectation.

Furthermore, as @yousong already pointed out, the `/var/etc/collectd.conf` symlink points to an ephemeral file which his recreated on every boot.

closed 2019-08-26 by jow-