openwrt/luci #3021: kmod-ledtrig-netdev: Can insert the xt_LED module but cannot use it

LoC-2, 2019-08-28:
I have a "BT Home Hub 5A" running the latest stable OpenWrt (18.06.4). The device has several lights on the front, most of which are controllable though the web interface and I'm trying to make them triggerable though an IPTables rule (see: https://openwrt.org/docs/guide-user/base-system/led_configuration).

So far I have:
* Run _opkg update_
* Installed the _kmod-ipt-led_ package.
* Verified via modprobe that _xt_LED_ is loaded
* Verified via _lsmod_ that _xt_LED_ is referring to _x_tables_

But my attempts to use & query the module produce an error:

````
root@OpenWrt:~# iptables -m led --help
iptables v1.6.2: Couldn't load match `led':No such file or directory

Try `iptables -h' or 'iptables --help' for more information.
````
````
root@OpenWrt:~# iptables -A INPUT -p tcp --dport 22 -j LED --led-trigger-id ssh --led-delay 1000
iptables v1.6.2: unknown option "--led-trigger-id"
Try `iptables -h' or 'iptables --help' for more information.
````
````
root@OpenWrt:~# iptables -A INPUT -p tcp --dport 22 -j LED --led-delay 1000
iptables v1.6.2: unknown option "--led-delay"
Try `iptables -h' or 'iptables --help' for more information.
````
````
root@OpenWrt:~# iptables -A INPUT -p tcp --dport 22 -j LED
iptables v1.6.2: Couldn't load target `LED':No such file or directory

Try `iptables -h' or 'iptables --help' for more information.
````

Additionally the target is not being listed in /proc/net/ip_tables_matches.

yousong, 2019-08-28:
> I have a "BT Home Hub 5A" running the latest stable OpenWrt (18.06.4). The
> device has several lights on the front, most of which are controllable
> though the web interface and I'm trying to make them [...]
Try also installing iptables-mod-led
Good day,
                yousong

closed 2019-08-28 by yousong

yousong, 2019-08-28:
The issue is from base system and please consider reporting to https://bugs.openwrt.org next time.  Thank you.

LoC-2, 2019-08-28:
Thanks. That worked.