openwrt/luci #3036: :depends doesn't work, adapt a list in function of another

erwan200026, 2019-09-03:
I would like my model list to be adapted in function of my brand list. Can you help me please?
If the Elsys brand is selected, I only want these models in the model list : ELT-1, ELT-2-HP, ELT-1-Lite, ERS-Eye, ERS Lite, ERS.
If the ATIM brand is selected, I only want these models in the model list : ACW-DINDIO, ACW-DINDIO.
Etc...

Here is my code:
`m=Map("lorawan-server","Manage LoRaWAN devices")`
`m.redirect = luci.dispatcher.build_url("admin/network/lorawan-server/devices-overview")`

`device=m:section(TypedSection,"device")`
`device.addremove=true`
`device.anonymous=false`
`device.template = "cbi/tblsection"`

`myListBrand=device:option(ListValue,"brand","Brand")`
`myListBrand:value("Elsys",translate("Elsys"))`
`myListBrand:value("Adeunis",translate("Adeunis"))`
`myListBrand:value("ATIM",translate("ATIM"))`

`myListModel=device:option(ListValue,"model","Model")`
`myListModel:value("ELT-1",translate("ELT-1")):depends("brand","Elsys")`
`myListModel:value("ELT-2-HP",translate("ELT-2-HP")):depends("brand","Elsys")`
`myListModel:value("ELT-1-Lite",translate("ELT-1-Lite")):depends("brand","Elsys")`
`myListModel:value("ERS-Eye",translate("ERS-Eye")):depends("brand","Elsys")`
`myListModel:value("ERS Lite",translate("ERS Lite")):depends("brand","Elsys")`
`myListModel:value("ERS",translate("ERS")):depends("brand","Elsys")`
`myListModel:value("Field Test Device",translate("Field Test Device")):depends("brand","Adeunis")`
`myListModel:value("ACW-DINDIO",translate("ACW-DINDIO")):depends("brand","ATIM")`
`myListModel:value("ACW-DINDA",translate("ACW-DINDIO")):depends("brand","ATIM")`

`return m`

jow-, 2019-09-03:
The correct syntax is `myListModel:value("ELT-1",translate("ELT-1"), {"brand"="Elsys"})`

closed 2019-09-03 by jow-

erwan200026, 2019-09-04:
Thank you so much !!!