openwrt/luci #3037: [bug] dump ap as dhcp client without dnsmasq -- no hostnames of clients

mistepien, 2019-09-03:
There is dump ap with br-lan interface configured as dhcp client. There is no dnsmasq installed on dump ap.

Router is fully functional and uses dnsmasq as a dns and dhcp server. All wireless clients get ip from  dnsmasq running on router. All of them send hostnames to dnsmasq running on router and the hostnames can be resolved via dns service provided by this dnsmasq.

Dump AP uses dns service provided by dnsmasq provide by router and may resolve local and remote hostnames.

However there are not hostnames on status luci page of dump ap. Instead of hostnames there are question-marks ("?").

jow-, 2019-09-04:
Are the hostnames present in /tmp/dhcp.leases? If not, then this is not a bug.

closed 2019-09-04 by jow-

mistepien, 2019-09-04:
If I copy dhcp.lease to dump AP then on luci status page I have informations about "Active DHCP Leases" and about "Associated Stations". But dump ap has nothing in common with "Active DHCP Leases". There is no dhcp server running on dump ap.

mistepien, 2019-09-04:
how to disable showing "Active DHCP Leases"?

Floriszz, 2026-01-04:
@Mistepien, did you ever found a solution to get the host names on the luci status page of the dumb Access Point?
I don’t have an Openwrt router, so no dhcp lease file available from router.
I would expect a simple ARP service. But after installing arp service my dumb Access Point crashed and had to restore from factory.