openwrt/luci #3051: JasonRpc Api does not work (using Release 18.06.4)

melaLetics, 2019-09-09:
Hey!

I've still got a problem consuming the JsonRpc api, also after installing a younger release than Chaos Calmer (see closed ticket 3050: https://github.com/openwrt/luci/issues/3050 )

I've now got a VM using OpenWrt 18.06.4 (r7808-ef686b7292). 
I use this virtualisation to play around with it before using the "real world application".
I've installed LuCi -> works fine.
I've installed luci-mod-rpc (git-19.241.48373-38f5f4b-1)  and restarted the uhttpd.

I'm able to get an auth token, sending with Postman a Post statement to
http://x.x.x.x/cgi-bin/luci/rpc/auth 
with json-data
{
"id": 1,
"method": "login",
"params": [
"root",
"xxxxxxx"
]
}

Fine, but using that token, I'm not able to get any response. No matter what library I try to reach.
Sadly, the library links in this Wiki are broken, so I have to Trial & Error :(

First question: do I send a Post or a Get statement? I've tried both, but any response...
So I tried:
http://x.x.x.x/cgi-bin/luci/rpc/fs?auth=f78bf762b59533713db182b45d7fb7e2
with json-data
{
"jsonrpc": "2.0",
"method": "dir",
"params": ["/"],
"id": 1
}

The response is
### Could not get any response
There was an error connecting to .

So how to get the jsonrpc work?
Is there anywhere a running documentation for the exported libraries? All links in https://github.com/openwrt/luci/wiki/JsonRpcHowTo are broken :(

Thanks in advance!

jacekpaszkowski, 2020-01-06:
@melaLetics Has your problem been solved?

Now i have similar issues with rpc api. It seems that nothing but auth method works...

http://192.168.1.1/cgi-bin/luci/rpc/fs
```
{ "jsonrpc": "2.0", "id": 2, "method": "dir", "params": ["/"] }
```
response:
```
{
    "id": null,
    "error": {
        "message": "Parse error.",
        "code": -32700
    },
    "jsonrpc": "2.0"
}
```

jacekpaszkowski, 2020-01-07:
@melaLetics my just Has been solved.
Take a look https://github.com/openwrt/luci/issues/3470

melaLetics, 2020-07-21:
Yeah, thanks. That worked for me!

closed 2020-07-21 by melaLetics