openwrt/luci #3054: WiFi edit issue after rewrite

MartB, 2019-09-11:
```
Cannot read property 'selected' of undefined
  at ClassConstructor.setValues (http://10.0.0.6/luci-static/resources/view/network/wireless.js:42:67)
  at ClassConstructor.toggleWifiChannel (http://10.0.0.6/luci-static/resources/view/network/wireless.js:43:648)
  at ClassConstructor.toggleWifiBand (http://10.0.0.6/luci-static/resources/view/network/wireless.js:43:507)
  at ClassConstructor.toggleWifiMode (http://10.0.0.6/luci-static/resources/view/network/wireless.js:43:175)
  at ClassConstructor.setInitialValues (http://10.0.0.6/luci-static/resources/view/network/wireless.js:46:20)
  at ClassConstructor.renderWidget (http://10.0.0.6/luci-static/resources/view/network/wireless.js:48:822)
  at async Promise.all (index 2)
  at async Promise.all (index 0)
  at async Promise.all (index 1)
```

wifi-config:
```
config wifi-device 'radio0'
        option type 'mac80211'
        option channel '36'
        option hwmode '11a'
        option path 'pci0000:00/0000:00:00.0'
        option htmode 'VHT80'

config wifi-device 'radio1'
        option type 'mac80211'
        option channel '11'
        option hwmode '11g'
        option path 'platform/ahb/18100000.wmac'
        option htmode 'HT20'

config wifi-iface 'wifinet0'
        option ssid 'test'
        option encryption 'none'
        option device 'radio0'
        option mode 'ap'
        option network 'lan'

config wifi-iface 'wifinet1'
        option ssid 'test'
        option encryption 'none'
        option device 'radio1'
        option mode 'ap'
        option network 'lan'
```

https://github.com/openwrt/luci/blob/fd40efdf4da6322fef9e3b02e713c914bc8ad530/modules/luci-mod-network/htdocs/luci-static/resources/view/network/wireless.js#L288

Am i missing something here or why does this happen?

MartB, 2019-09-11:
Okay nvm,
After checking https://github.com/openwrt/luci/commit/963b7636b46fd1b9ae47e2317ef491ce1b0ecfd5 installing `rpcd-mod-iwinfo` fixed it for me.

Leaving this open for others to see, close it if you feel like it.

jow-, 2019-09-14:
Thanks for confirming the fix. Fixed in f32f42bc259c2c24f4369e945123a1f06b33b0a3

closed 2019-09-14 by jow-