openwrt/luci #3068: Client Id to send When requesting DHCP defaults to 'root'

RobGray-au, 2019-09-15:
causes error with some DHCP servers (including NetGear and my ISP)
wan is setting up now
udhcp: malformed hex string 'root'

the fix was pointed out by [eduperez] WRT3200ACM no connect to ISP ... but i didnt know where to find setting in LuCi

for those coming later....
Network
Interfaces
WAN

Client Id to send When requesting DHCP  _swap_ root _with_ A1 _or other HEXstring_
<save & Apply>

dhcp connected almost immediately.

jow-, 2019-09-16:
This is a browser bug, LuCI does not perform any kind of autocomplete.

closed 2019-09-16 by jow-

RobGray-au, 2019-09-16:
I dont believe it is either browser or LuCi.

from [https://openwrt.org/docs/guide-user/network/ipv4/start#protocol_dhcp](  )
clientid | string | no | _system default_ | Override client identifier in DHCP requests

it appears the system default is _root_ , where some (presumably poorly implemented) DHCP servers expect a MAC address, or any valid Hex string. ie F2 works.
note that LuCi wont allow a blank ClientId, it reverts to [root]

I couldnt find this hack/workaround easily, so perhaps hope this helps others to do so now.
this issue hasnt raised for me previosly as the device was set with a static IP when acting as access point only.  same output on a different hardware running 18.06.2

jow-, 2019-09-16:
There is no `root` default for client id, as can be seen here: https://github.com/openwrt/openwrt/blob/8852a09d85e5b0c2ffab55578a9624edb2fb6dd1/package/network/config/netifd/files/lib/netifd/proto/dhcp.sh#L54

If not set, `-C` is passed to `udhcpc` which instructs it to not set any ID at all. The `root` value is added by some browsers because they confuse the interface setting with a login form and helpfully add nonsense into the input field. It is certainly not LuCI which adds `root` as value.