openwrt/luci #3072: luci-app-statistics: ping plugin does not support IPv6 numeric address

dengqf6, 2019-09-15:
When I set the _Monitor hosts_ of ping plugin to an IPv6 numeric address such as `2001:470:20::2`, the graph will not load.
Then I got these error messages in system log:
```
daemon.err uhttpd[1581]: ERROR: can't parse DEF '1ping_avg_raw=/tmp/rrd/OpenWrt/ping/ping-2001\\:470\\:20\\:\\:2.rrd:value:AVERAGE' -2
daemon.err uhttpd[1581]: ERROR: can't parse DEF '1ping_droprate_avg_raw=/tmp/rrd/OpenWrt/ping/ping_droprate-2001\\:470\\:20\\:\\:2.rrd:value:AVERAGE' -2
```

feckert, 2019-09-16:
This should work in the master and openwrt-19.06 branch.
The change was added with the following commit.
https://github.com/openwrt/luci/commit/7bab4a3ef23397991c7f2296508cd52be20b4e49#diff-64c154d89ea097b12be97c2bc71b09da

dengqf6, 2019-09-16:
I'm using master

feckert, 2019-09-19:
But it looks more like the data doesn't match. I think the file can already be opened

adelton, 2025-01-10:
On 23.05.5 with luci-app-statistics git-24.364.71483-75d2b84 and collectd-mod-ping 5.12.0-49, graphs with IPv6 addresses seem to work:
[image]

Can this issue be marked as resolved?

closed 2025-01-10 by hnyman