openwrt/luci #3076: QMI interfaces do not display  RX TX count

xewonder, 2019-09-17:
Hello,

OpenWrt  latest pull  (as of now) and on interfaces QMI does not display the RX / TX bytes count.

Mouse over the icon, display the RX / TX
[image]
[image]

Any help or fix apreciated.

Thank you very much

jow-, 2019-09-17:
Please provide the output of `ubus call network.interface dump` and `ubus call network.device status`

xewonder, 2019-09-17:
root@OpenWrt:~# ubus call network.interface dump
{
        "interface": [
                {
                        "interface": "4G",
                        "up": false,
                        "pending": false,
                        "available": false,
                        "autostart": false,
                        "dynamic": false,
                        "proto": "3g",
                        "data": {

                        }
                },
                {
                        "interface": "lan",
                        "up": true,
                        "pending": false,
                        "available": true,
                        "autostart": true,
                        "dynamic": false,
                        "uptime": 357,
                        "l3_device": "br-lan",
                        "proto": "static",
                        "device": "br-lan",
                        "updated": [
                                "addresses"
                        ],
                        "metric": 0,
                        "dns_metric": 0,
                        "delegation": true,
                        "ipv4-address": [
                                {
                                        "address": "192.168.1.1",
                                        "mask": 24
                                }
                        ],
                        "ipv6-address": [

                        ],
                        "ipv6-prefix": [

                        ],
                        "ipv6-prefix-assignment": [
                                {
                                        "address": "fde9:80f9:259a::",
                                        "mask": 60,
                                        "local-address": {
                                                "address": "fde9:80f9:259a::1",
                                                "mask": 60
                                        }
                                }
                        ],
                        "route": [

                        ],
                        "dns-server": [

                        ],
                        "dns-search": [

                        ],
                        "neighbors": [

                        ],
                        "inactive": {
                                "ipv4-address": [

                                ],
[... 246 lines cut ...]
                        "pending": true,
                        "available": true,
                        "autostart": true,
                        "dynamic": false,
                        "proto": "dhcp",
                        "device": "eth0.2",
                        "data": {

                        }
                },
                {
                        "interface": "wan6",
                        "up": false,
                        "pending": true,
                        "available": true,
                        "autostart": true,
                        "dynamic": false,
                        "proto": "dhcpv6",
                        "device": "eth0.2",
                        "data": {

                        }
                }
        ]
}

xewonder, 2019-09-17:
root@OpenWrt:~# ubus call network.interface.sim2 status
{
        "up": true,
        "pending": false,
        "available": true,
        "autostart": true,
        "dynamic": false,
        "uptime": 275,
        "l3_device": "wwan0",
        "proto": "qmi",
        "updated": [
                "data"
        ],
        "metric": 0,
        "dns_metric": 0,
        "delegation": true,
        "ipv4-address": [

        ],
        "ipv6-address": [

        ],
        "ipv6-prefix": [

        ],
        "ipv6-prefix-assignment": [

        ],
        "route": [

        ],
        "dns-server": [

        ],
        "dns-search": [

        ],
        "neighbors": [

        ],
        "inactive": {
                "ipv4-address": [

                ],
                "ipv6-address": [

                ],
                "route": [

                ],
                "dns-server": [

                ],
                "dns-search": [

                ],
                "neighbors": [

                ]
        },
        "data": {
                "cid_4": "20",
                "pdh_4": "-2030522560"
        }
}

xewonder, 2019-09-17:
... Thank you for your attention..

closed 2019-09-18 by jow-

xewonder, 2019-09-18:
hi, 
I applied the "patch" but id does not change anything...

[image]

Thank you

jow-, 2019-09-18:
Make sure to clear your browser cache.

xewonder, 2019-09-18:
oh dear! you are right agin! sorry about that....
THANK YOU!

commit 15cc74c6 (2019-09-18): luci-mod-network: interfaces.js: derive status from layer 3 device

Align the status column device selection logic with the approach used for
rendering the interface status boxes; first try to obtain the active
layer3 device as reported by ubus, only then fall back to the protocol
specific device hint which might or might not point to an existing device.

Fixes: #3076