openwrt/luci #3078: Wireless - Icon is crossed if no LAN device is assigned

xewonder, 2019-09-17:
Hello,

OpenWrt latest pull (as of now) and if an SSID does not have a network (Valid) device assigned to it, it is "crossed" even though operational...

[image]

Thank you in advance

jow-, 2019-09-17:
Please provide the output of `ubus call network.wireless status`

xewonder, 2019-09-17:
Hi,

please note that "hotspot" i a non-existant interface assigned with uci set at boot.

root@OpenWrt:~# ubus call network.wireless status
{
        "radio0": {
                "up": true,
                "pending": false,
                "autostart": true,
                "disabled": false,
                "retry_setup_failed": false,
                "config": {
                        "hwmode": "11g",
                        "path": "platform/10180000.wmac",
                        "country": "00",
                        "channel": "7",
                        "htmode": "HT40"
                },
                "interfaces": [
                        {
                                "section": "default_radio0",
                                "ifname": "wlan0",
                                "config": {
                                        "mode": "ap",
                                        "encryption": "none",
                                        "wmm": false,
                                        "ssid": "Waveloc Public",
                                        "mode": "ap",
                                        "network": [
                                                "hotspot"
                                        ]
                                }
                        },
                        {
                                "section": "wifinet0",
                                "ifname": "wlan0-1",
                                "config": {
                                        "encryption": "none",
                                        "mode": "ap",
                                        "ssid": "Waveloc2",
                                        "mode": "ap",
                                        "network": [
                                                "lan"
                                        ]
                                }
                        }
                ]
        }
}

Thank you

commit 10880b56 (2019-09-18): luci-base: network.js: get low level device details from getifaddr() info

Fixes: #3078
closed 2019-09-18 by jow-