openwrt/luci #3079: Missing dependencies on Snapshot 19.07

dmitrygribenchuk, 2019-09-17:
Hello! I'm try to build OpenWrt on Snapshot 19.07 and have problem with satisfy dependencies.

Collected errors:
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for luci:
 * 	libip4tc2
 * 	libip6tc2

hnyman, 2019-09-17:
You need to clean your build environment. There are probably previously compiled packages that have slightly different versions, causing the dependency to fail.

Do "make clean" before the build.

(I just built 19.07 without problems...)

closed 2019-09-17 by hnyman

dmitrygribenchuk, 2019-09-17:
No. It's doesn't work

strasharo, 2019-12-26:
Currently hitting the same with https://downloads.openwrt.org/snapshots/targets/ar71xx/tiny/openwrt-imagebuilder-ar71xx-tiny.Linux-x86_64.tar.xz
Has anyone found a solution for this?

dmitrygribenchuk, 2019-12-27:
> Currently hitting the same with https://downloads.openwrt.org/snapshots/targets/ar71xx/tiny/openwrt-imagebuilder-ar71xx-tiny.Linux-x86_64.tar.xz
> Has anyone found a solution for this?

Yep. My device was moved from ar71xx to ath79 and change profile name (from tl-wr841-v8 to tplink_tl-wr841-v8). Check it.

strasharo, 2019-12-27:
Thanks a lot @dmitrygribenchuk . Switched to https://downloads.openwrt.org/snapshots/targets/ath79/tiny/openwrt-imagebuilder-ath79-tiny.Linux-x86_64.tar.xz and profile from tl-wr703n-v1 to tplink_tl-wr703n and that dit it for me. Cheers!