openwrt/luci #3084: Luci - Network - Interfaces - WireGuard TX/RX stats not working

dk4dk4, 2019-09-20:
I recently compiled my OpenWRT build to OpenWRT SNAPSHOT r11033-2b342d01a2 
I custom packaged WireGuard via Makefile to latest ver 0.0.20190913-1

The WireGuard interface no longer counts the RX/TX bytes properly:
[image]

When the Status - WireGuard Status page reports:
[image]

The other non-wireguard interfaces continue to count RX/TX properly.  It is just the new WG ver

jow-, 2019-09-20:
Please provide the output of `ubus call network.interface dump`

dk4dk4, 2019-09-20:
root@DKWrt:~# ubus call network.interface dump
{
        "interface": [
                {
                        "interface": "ipvanvpntun",
                        "up": true,
                        "pending": false,
                        "available": true,
                        "autostart": true,
                        "dynamic": false,
                        "uptime": 28406,
                        "l3_device": "tun0",
                        "proto": "none",
                        "device": "tun0",
                        "metric": 0,
                        "dns_metric": 0,
                        "delegation": true,
                        "ipv4-address": [

                        ],
                        "ipv6-address": [

                        ],
                        "ipv6-prefix": [

                        ],
                        "ipv6-prefix-assignment": [

                        ],
                        "route": [

                        ],
                        "dns-server": [

                        ],
                        "dns-search": [

                        ],
                        "neighbors": [

                        ],
                        "inactive": {
                                "ipv4-address": [

                                ],
                                "ipv6-address": [

                                ],
                                "route": [

                                ],
                                "dns-server": [

                                ],
                                "dns-search": [

                                ],
                                "neighbors": [

                                ]
                        },
                        "data": {

                        }
                },
                {
                        "interface": "lan",
                        "up": true,
                        "pending": false,
                        "available": true,
[... 238 lines cut ...]
                        "route": [
                                {
                                        "target": "0.0.0.0",
                                        "mask": 0,
                                        "nexthop": "24.xx.xx.1",
                                        "source": "24.xx.xx.79/32"
                                }
                        ],
                        "dns-server": [
                                "1.0.0.1",
                                "1.1.1.1"
                        ],
                        "dns-search": [

                        ],
                        "neighbors": [

                        ],
                        "inactive": {
                                "ipv4-address": [

                                ],
                                "ipv6-address": [

                                ],

jow-, 2019-09-20:
What is the version of LuCI installed?

closed 2019-09-20 by jow-