openwrt/luci #3086: Luci -- dnsmasq + proxydnssec

mistepien, 2019-09-20:
It would be great to have a swich in Luci to deal with uci option "dhcp.@dnsmasq[0].proxydnssec".

jow-, 2019-09-20:
What is the label and description of `proxydnssec`?
What are its possible values?
Does it depend on other options?

mistepien, 2019-09-20:
in /etc/init.d/dnsmasq there is a line
`append_bool "$cfg" proxydnssec "--proxy-dnssec`
thus this is bool, and think it should not be turn on together with dnssec enabled because with proxydnssec dnsmasq does not validate dnssec by itself.

this is what is manual:
> --proxy-dnssec Copy the DNSSEC Authenticated Data bit from upstream servers to downstream clients and cache it. This is an alternative to having dnsmasq validate DNSSEC, but it depends on the security of the network between dnsmasq and the upstream servers, and the trustworthiness of the upstream servers.

This is useful in case of using stubby/unbound + dnsmasq
https://forum.openwrt.org/t/tutorial-dns-over-tls-with-dnsmasq-and-stubby-no-need-for-unbound/18663/16

mistepien, 2019-09-27:
I found that `proxydnssec` works with no-DNSSEC dnsmasq, too. Thus dnsmasq does not need to be dnsmasq-full.

ne20002, 2019-12-09:
I'd love to have this feature ... and to use an unbound in front of the included dnsmasq, basing the dnssec validation on the unboud (which i don trust, because it is mine).
But from my tests with pi-hole (which is based on dnsmasq) I see the proxy-dnssec is not working the way one would expect.
If I do a first query with dig I get the ad flag delivered by a dnsmasq with proxy-dnssec set. But any subsequent dig calls don't deliver the ad flag, as it seems it is **not** chached by dnsmasq. 
If this is the case for the dnsmasq in OpenWrt as well and not a problem of the pi-hole version, this 'feature' is useless.