openwrt/luci #3090: openwrt-theme: XML parsing error

dibdot, 2019-09-21:
Fresh image from today, System => Startup is broken in standard theme, e.g.

[image]

jow-, 2019-09-21:
Which LuCI version?

dibdot, 2019-09-21:
OpenWrt SNAPSHOT r11083-8cb13f4e6d / LuCI Master (git-19.263.54698-720df98)

jow-, 2019-09-21:
Hmm, can you check your browser debug console and see if you can obtain a backtrace? I cannot reproduce it here so far. The error is thrown by the `E()` function when some invalid (X)HTML snippets are passed to it. This used to be the case for `Save & Apply` (which must be `Save &amp; Apply` for XHTML) but I fixed these particular cases a few commits back, so something else specific to your setup must be triggering it.

dibdot, 2019-09-21:
OK, I found the root cause, it depends on special chars in the "Local Startup" (rc.local) textarea, e.g. this breaks the XML-view:
```
# re-enable & start remote ssh tunnel
#
if [ $(/etc/init.d/ssh-tunnel enabled; printf "%u" ${?}) -eq 1 ]
then
    /etc/init.d/ssh-tunnel enable
    /etc/init.d/ssh-tunnel start
fi
```

feckert, 2019-09-23:
@dibdot i think this could be closed

jow-, 2019-09-23:
@feckert - nope, we need to fix it. @dibdot thanks for the info.

feckert, 2019-09-23:
Ok understood I thought this had to do with his adapted rc.local
But this is then a general problem and not only for the luci-theme-openwrt.
So we should change the header to` luci-base: ...`

jow-, 2019-09-23:
Its actually a problem with the startup and crontab pages specifically as these manually construct HTML containing `<textarea>` elements.

commit f3ff641d (2019-09-23): luci-mod-system: ensure that textarea contents are properly escaped

Fixes: #3090
closed 2019-09-23 by jow-