openwrt/luci #3092: [RFC] Use current master for 19.07?

jow-, 2019-09-21:
I've been thinking about merging the current master state back into the `openwrt-19.07` branch since LuCI muster now more or less has the feature set I wanted for `19.07` .

Whats your opinions on this? In particular, are all custom LuCI apps still working correctly and can they simply be backported to 19.07 or does that require changes in the packages repo? Are e.g. adblock master and adblock 19.07 compatible to each other?

The biggest advantage I see is that we'll regain the ability to cherry-pick fixes back into 19.07 which right now is completely different to master.

On the downside, LuCI will likely get slightly larger due to redundant JS and Lua implementations for some classes - but this overhead should get less over time as more things are ported to client side functionality.

@hnyman @dibdot @stangri

hnyman, 2019-09-21:
I feel that it might be ok. The 19.07 branch has been "on hold" for several months while LuCI has progressed significantly, e.g. with WPA3 compatilibity etc.

Backporting the whole LuCI master to 19.07 might require some changes (or at least verification) to wifi components in core OpenWrt, as iwinfo, hostapd etc. have been enhanced to support wpa3 better, and I am not sure that all corresponding changes have been backported, yet.

Regarding packages, I have tried to keep 19.07 and master in sync for most of the packages that I am active with. ( @feckert has recently modified collectd and luci-statistics, so the collectd changes should be backported to 19.07. )

As far as I know, @stangri and @dibdot have their master packages pretty much backported to 19.07

feckert, 2019-09-21:
The first thing that occurs to me is that the changes seem to have been tested with the bootstrap theme only. I installed the material theme once and the modal looks broken. I have a fix in my patchset for this I will publish if I finished the tests.

>  @feckert has recently modified collectd and luci-statistics, so the collectd changes should be backported to 19.07.

I was also in the process of changing the luci-app-statistics to use the collectd. Have a look at the following branch I already uploaded to. https://github.com/TDT-AG/luci/tree/pr/20190917-luci-app-statistics but did not requested a pullrequest because this is work in progress.

But I must say I like the approach with the Javascript this makes the LuCI faster. But it will delay the release for the 19.07 a little bit, because there will be some issues raise up.

dibdot, 2019-09-21:
@jow- please go forward and merge the current master in 19.07 ... I like the new JS implementation a lot. Adblock, banIP and Travelmate are fully compatible with current master and already backported to 19.07 branch as well.

hnyman, 2019-10-01:
@jow- 
Relating backporting LuCI, also the underlying wifi components really need backporting (iwinfo, hostapd etc.). I tried a few days ago using the mixed wpa2/wpa3 config and while it worked in master, the same config did not work with 19.07. (wifi clients failed to get connected)

closed 2019-11-03 by jow-