openwrt/luci #3098: Luci - Network - Interfaces - WG - restart broken

dk4dk4, 2019-09-22:
Powered by LuCI Master (git-19.263.54698-720df98) / OpenWRT SNAPSHOT r11070-c5b10c8282
WireGuard via Makefile to latest ver 0.0.20190913-1

I'm no longer able to restart a WireGuard interface using the Luci Interface page:

[image]

If I reboot the router the interface comes up no problem.  
When I click restart it fails to restart the interface, and when I check under Status - WireGuard status I get the following error:
[image]

The error says there is no public key but the connection works fine when I reboot?  Something is off there...

jow-, 2019-09-22:
The `Restart` button simply executes `/sbin/ifup mullvad` behind the scenes. Is this working fine when executed via the cli?

dk4dk4, 2019-09-22:
The cli produces the same results.  The interface does not come back up.  When I check wireguard status page it has the same "Interface does not have a  public key" message.  When I rebooted it came back up ok...

jow-, 2019-09-22:
This isn't a LuCI bug then but an issue with the wg protocol handler implementation. I didn't track if there's been any recent changes to it.

dk4dk4, 2019-09-22:
Yes I wasn't sure where to open the ticket really. Can you move it to the
appropriate spot ?

jow-, 2019-09-22:
Lets leave it open for the time being, I'll see if I can take a look at it during the next days.

dibdot, 2019-09-22:
In OpenWrt is currently wireguard 0.0.20190702 ... I can't reproduce such restart errors with that version.

dk4dk4, 2019-09-22:
Correct. I compiled to latest WRT and WG builds and this is new behaviour.
I listed the versions at start.

dk4dk4, 2019-10-08:
I recently completed another build to:
OpenWRT SNAPSHOT r11159
The Wireguard included was the same ver as I had built before 0.0.20190913-1
WG interface issue persists.  I'm not able to restart the WG interface without the error.  Reboot resolves and WG connects successfully.

dk4dk4, 2019-10-24:
This seems to be resolved with the latest WG bump to ver 0.0.20191012-1.
I can restart the WG interface again via Luci.

Rebuilt and tested with:
Powered by LuCI Master (git-19.296.33125-a9b680c) / OpenWRT SNAPSHOT r11320-2bc7c519dc

closed 2019-10-24 by dk4dk4