openwrt/luci #3100: Wireless scan shows WPA2 instead of WPA3 and NONE instead of OWE or SAE

IronicSven, 2019-09-22:
I've build openwrt with wpad-wolfssl and luci from master and configured different WPA3 modes via luci on all of my devices.

A wireless scan of the other devices in luci displays in the encryption column
**WPA2 PSK (CCMP)** instead of **mixed WPA2/WPA3 PSK, SAE (CCMP)**
and
**WPA2 NONE (CCMP)** instead of **WPA3 SAE (CCMP)**
and
**WPA2 NONE (CCMP)** instead of **WPA3 OWE (CCMP)**

jow-, 2019-09-22:
Please provide the output of `opkg list-installed | grep iwinfo`

hnyman, 2019-09-22:
Are you talking about master or 19.07?

(the mixed mode showed me ok in master, yesterday).

IronicSven, 2019-09-22:
opkg output:
```
 OpenWrt SNAPSHOT, r11094-a03219ba09
 -----------------------------------------------------
root@ArcherC7v2:~# opkg list-installed | grep iwinfo
iwinfo - 2019-09-20-02112f9b-1
libiwinfo-lua - 2019-09-20-02112f9b-1
libiwinfo20181126 - 2019-09-20-02112f9b-1
rpcd-mod-iwinfo - 2019-09-21-95f0973c-1
```
iwinfo wlan1 scan on Archer C7 v2:
```
root@ArcherC7v2:~# iwinfo wlan1 scan
...
Cell 17 - Address: 70:4F:57:45:49:D4
          ESSID: "StPauliNet"
          Mode: Master  Channel: 6
          Signal: -85 dBm  Quality: 25/70
          Encryption: WPA2 PSK (CCMP)
```

iwinfo on 1043ND v5:
```
 OpenWrt SNAPSHOT, r11094-a03219ba09
 -----------------------------------------------------
root@1043NDv5:~# iwinfo
wlan0     ESSID: "StPauliNet"
          Access Point: 70:4F:57:45:49:D4
          Mode: Master  Channel: 6 (2.437 GHz)
          Tx-Power: 23 dBm  Link Quality: 66/70
          Signal: -44 dBm  Noise: -95 dBm
          Bit Rate: 97.6 MBit/s
          Encryption: mixed WPA2/WPA3 PSK/SAE (CCMP)
          Type: nl80211  HW Mode(s): 802.11bgn
          Hardware: unknown [Generic MAC80211]
          TX power offset: unknown
          Frequency offset: unknown
          Supports VAPs: yes  PHY name: phy0
```

Does this mean the issue is rather **iwinfo <device> scan** related than a luci problem?

jow-, 2019-09-22:
Whats the output of `iwinfo wlan1 info` on the Archer?

IronicSven, 2019-09-22:
```
root@ArcherC7v2:~# iwinfo wlan1 info
wlan1     ESSID: "FlensNet"
          Access Point: C4:6E:1F:EA:C1:52
          Mode: Master  Channel: 11 (2.462 GHz)
          Tx-Power: 24 dBm  Link Quality: unknown/70
          Signal: unknown  Noise: -94 dBm
          Bit Rate: unknown
          Encryption: mixed WPA2/WPA3 PSK/SAE (CCMP)
          Type: nl80211  HW Mode(s): 802.11bgn
          Hardware: unknown [Generic MAC80211]
          TX power offset: unknown
          Frequency offset: unknown
          Supports VAPs: yes  PHY name: phy1
```

jow-, 2019-09-22:
Thanks, I can reproduce it. Its an libiwinfo issue. Right now it only recognized WPA3 when scanning through wpa_supplicant, not when scanning from netlink directly.

jow-, 2019-09-22:
Fixed with https://github.com/openwrt/openwrt/commit/ced4c0e6352a7994f4facd3de2e541ba073a7008

closed 2019-09-22 by jow-

IronicSven, 2019-09-22:
Thank you for your great work! Everything is displayed correctly now.