openwrt/luci #3101: luci does not allow to change MAC number of wan interface

mistepien, 2019-09-22:
OpenWrt SNAPSHOT, r11070-c5b10c8282
luci has an option "Override MAC address" (in Advanced Settings of the interface). But it does not work.
luci sets "network.wan.macaddr", and does not touch "network.wan_dev.macaddr". The later options matters most.

jow-, 2019-09-23:
Dupe of #1100

closed 2019-09-23 by jow-

stman, 2020-05-20:
I have the same issue. I found why it is not working properly. I am using a Ubiquiti EdgeRouter X that incorporate an embedded Switch.

OpenWRT creates two internal ethernet devices for the LAN and WAN physical ethernet ports that are not shown by LUCI but that can be displayed using CLI command "uci show network". Internaly, the switch is connected to the original LAN port.

If you use UCI commands to alter the Mac address of these two internal and not shown in LUCI ethernet ports, and then do some tcpdump on those interfaces, you will see that the change of mac address works, but it is still not display correctly by LUCI.

Indeed, for routers with embedded switch, LUCI doesn't manage well these two exceptions for the two physical ethernet ports.