openwrt/luci #3105: luci-i18n-base-es: "None" does not appear translated.

castillofrancodamian, 2019-09-23:
[image]

castillofrancodamian, 2019-10-01:
I have a Master. That is why I say that it is already translated, but not when I choose the translated language.

feckert, 2019-11-14:
I think that's an established term.
And therefore needs no translation.

urbalazs, 2019-12-22:
"None" is translatable and available for translation on Weblate. As I see, it is already translated into Spanish. This issue can be closed.

castillofrancodamian, 2019-12-23:
Yes, I know it is translated. The problem is that it is not displayed in LuCI.

feckert, 2021-09-03:
I have now taken a closer look.
This cannot be translated, because it is the uci value of the encryption option

`option encryption 'none'`

closed 2021-09-03 by feckert