openwrt/luci #3108: Luci Broken in Master ar71xx

castillofrancodamian, 2019-09-24:
**Relevant data:**

Device: TP-Link CPE210v2

Image: http://downloads.openwrt.org/snapshots/targets/ar71xx/generic/openwrt-ar71xx-generic-cpe210-v2-squashfs-sysupgrade.bin

`opkg update && opkg install luci`

[image]

It also happens in Archer C60v1

jow-, 2019-09-24:
ar71xx snapshots are discontinued, use ath79 ones instead.

closed 2019-09-24 by jow-

castillofrancodamian, 2019-09-24:
I just installed Archer C60 ath79 and the same thing happens.

jow-, 2019-09-24:
I cannot reproduce this, make sure the `rpcd` process is running.

castillofrancodamian, 2019-09-28:
What should i do with `rpcd`?