openwrt/luci #3111: Wireless: limit allowed channels for regulatory reasons

renne, 2019-09-24:
Hi,

for regulatory reasons we need to limit the 5 GHz channels to 100 - 140.

Please add the hostap statement `chanlist` to L(UCI).

jow-, 2019-09-24:
This is too vague to act upon. What exactly do you want added where? What is the intended functionality? What are the labels and descriptions? What are the dependencies to other options?

renne, 2019-09-25:
e.g.  Germany:

| Channel | Frequency (GHz) | Allowed EIRP (mW) | Outdoor |
| ---: | :---: | ---: | :--- |
| 36 | 5,180 | 200 | No |
| 40 | 5,200 | 200 | No |
| 44 | 5,220 | 200 | No |
| 48 | 5,240 | 200 | No |
| 52 | 5,260 | 200 | No |
| 56 | 5,280 | 200 | No |
| 60 | 5,300 | 200 | No |
| 64 | 5,320 | 200 | No |
| 100 | 5,500 | 1000 | Yes |
| 104 | 5,520 | 1000 | Yes |
| 108 | 5,540 | 1000 | Yes |
| 112 | 5,560 | 1000 | Yes |
| 116 | 5,580 | 1000 | Yes |
| 120 | 5,600 | 1000 | Yes |
| 124 | 5,620 | 1000 | Yes |
| 128 | 5,640 | 1000 | Yes |
| 132 | 5,660 | 1000 | Yes |
| 136 | 5,680 | 1000 | Yes |
| 140 | 5,700 | 1000 | Yes |
| 149 | 5,745 | 25 | Yes |
| 153 | 5,765 | 25 | Yes |
| 157 | 5,785 | 25 | Yes |
| 161 | 5,805 | 25 | Yes |
| 165 | 5,825 | 25 | Yes |
| 155-171 | | 4000 | Commercially licensed services only |

The `auto` mode selects all channels between 36 and 149 which means you can only use 25 mW in `auto` mode.

The `hostapd` option `chanlist` allows to define allowed channels in auto mode (e.g. `36-64` or `100-140` or a list of channels e.g. `120,128,136`). We would e.g. need to limit the operation to channels 100-140 for indoor/outdoor operation with 1000 mW EIRP.

I suggest to add an option `Allowed channels` in LUCI for 5 Ghz operation which converts to `chanlist` in the `hostapd` configuration.

The only dependency is to 5 GHz (802.11a/h/j/n/ac) mode.

CyberHead, 2021-04-17:
👍

Ansuel, 2021-12-16:
@jow- some time ago I sent a patch to iwinfo to add support for it...
The idea is to give to ubus more data and the gui will alert the user that an outdoor channel is used. The user will then decide to use it or not.
We can't enforce this just as we can't enforce a correct reg domain set so I think an alert should be handy to support this.

Can you give it a look? There were some comments about some spelling mistake/implementation error but I would like to have a more general review if the idea is correct.

NilsRo, 2022-09-03:
Also needed for DFS channel selection. Without hostap cannot jump to another channel on radar detection which switches of Wifi

        option channel 'auto'
        option channels '136, 140, 132, 116, 112, 108, 104, 100'

https://github.com/openwrt/openwrt/issues/9864