openwrt/luci #3115: luci-proto-modemmanager not registered

nickberry17, 2019-09-25:
It looks like the luci-proto's have been revamped.  Can you explain what I need to do to make [luci-proto-modemmanager](https://github.com/nickberry17/luci-proto-modemmanager) compatible?

feckert, 2019-09-25:
You have to switch to javascript. 
See https://github.com/openwrt/luci/tree/master/protocols and 
https://github.com/openwrt/luci/tree/master/modules/luci-mod-network/htdocs/luci-static/resources/view/network

jow-, 2019-09-25:
I'll take a look into it tomorrow morning.

nickberry17, 2019-10-08:
> I'll take a look into it tomorrow morning.

Hey, if you can point me in the right direction I can probably get it started.  The ModemManager package [is now in master](https://github.com/openwrt/packages/commit/d3012ecf08770171e02e46db38ff6e39d82ba4b5).

feckert, 2019-10-08:
> You have to switch to javascript.
> See https://github.com/openwrt/luci/tree/master/protocols and
> https://github.com/openwrt/luci/tree/master/modules/luci-mod-network/htdocs/luci-static/resources/view/network

@nickberry17  Thats the right direction you are looking for :-)

closed 2019-10-11 by nickberry17