openwrt/luci #3118: LuCI openwrt-19.07: "Override MTU" Field Auto-Populates with Saved Login Username "root"

eginnc, 2019-09-26:
On an EA6350v3 running OpenWrt 19.07-SNAPSHOT r10554-26c0bec13b / LuCI openwrt-19.07 branch (git-19.263.64892-befe994), in the LuCI Interfaces>Common Configuration>Advanced Settings, the "Override MTU" field is populating with the login username "root" saved in my Firefox browser (69.0.1 running on Manjaro linux). Clearing my saved login in Firefox clears the problem - its just an annoyance, but one not present in 18.06.4 on my EdgeRouter-X or EA8500.

S2-, 2020-02-17:
I have this too in `Use gateway metric`
[image]
Probably a firefox bug, but this happened to me only in luci :(

adelton, 2025-01-10:
I'm not able to reproduce with OpenWrt 23.05.5 and Firefox 133. I assume something about the field order in 19.07 confused Firefox back then.

Do you still see the issue even with latest OpenWrt and browser versions?

systemcrash, 2025-01-10:
Probably faulty hinting.

closed 2025-01-10 by systemcrash